ShikOfTheRa/scarab-osd #170: Read current profile in MSP_STATUS

axyspg, 2016-01-04:
I'm working on MSP to SmartPort adapter heavily based on your code, good work by the way, and I noticed that Cleanflight sends the current profile number on the MSP_STATUS message, which I'm using in my adapter. If you want to take a peek on what I'm working on, you can check at http://someonedidthatalready.blogspot.com

Back to the issue, I'm not sure if the value left on buffer is discarded or may interfere, but here it is. Hope is useful (it's last read8())

```
  if (cmdMSP==MSP_STATUS)
  {
    cycleTime=read16();
    I2CError=read16();
    MwSensorPresent = read16();
    MwSensorActive = read32();
    #if defined FORCESENSORS
      MwSensorPresent=GPSSENSOR|BAROMETER|MAGNETOMETER|ACCELEROMETER;
    #endif  
    armed = (MwSensorActive & mode.armed) != 0;
    read8(); // skip profile number
  }
```

ShikOfTheRa, 2016-01-05:
Hi thanks for info - someone was asking for this so it may be of use.
unread buffer values are in effect discarded for reference!

ShikOfTheRa, 2016-01-06:
Added display of cleanflight profile number to roadmap wishlist

closed 2016-01-06 by ShikOfTheRa