ShikOfTheRa/scarab-osd #173: NTSC/PAL switching

btimby, 2016-01-11:
Hi, I had a problem the other day with my OSD. I had configured it for NTSC, while my camera was PAL. What happened is that I could see the video feed, but no overlay. If I unplugged the camera, I could see the overlay, but no video.

My problem was that I tried to use the OSD built-in menu system to switch to PAL. I could do this, but the display started rapidly scrolling such that I could not use the menu to save the setting. Power cycling reverted to the NTSC value and I had to try again. I was eventually able to do this blindly after a few tries and could then plug my camera back in.

My question is if there is a possibility to modify how toggling this setting is handled. I see two possibilities.

1. Change this option without applying it. The user could then safely save the changes before their video became unreadable. Upon reboot the new setting would be applied.

2. Save and apply this setting immediately so that the display becomes unreadable until the user reboots with the new setting enabled.

I am willing to submit a pull request for this change if you let me know if this is desirable and any comments you may have about this topic.

btimby, 2016-01-11:
I just found AUTOCAM by reading old issues, had no idea this existed! I still think the above change could be useful.

ShikOfTheRa, 2016-01-11:
Actually I'd like to make Autocam default - be haven't had enough real world testing of it. 

Maybe, add a line in to say if cameramode!oldcameramode  MAX7456Setup();
Code space is at a premium so best if can be defined out.... OK to leave on a s default for now. As new code is added we have to make a call over functionality that is enabled as default

This should force max chip to be re-initialised with new settings. Need to be tested though.

btimby, 2016-01-11:
I can do that! I have the perfect combo of components to test with ;-).

btimby, 2016-01-11:
I found this:

https://github.com/ShikOfTheRa/scarab-osd/blob/master/MW_OSD/Serial.ino#L720

I think what is happening is that the max chip is resetting to the new signal type, but the rest of the video stack is not (vtx, monitor etc) since these items generally auto-detect, and probably do so on power up.

Thus, I think, this call to `MAX7456Setup()` should be removed. My thinking is that if the user is in the menu, their video is working (at least partially) and they should be able to save the changes. In my situation as soon as I changed this value my display went crazy and I could not save the changes.

Think of changing your monitor settings, the display is changed immediately, but if you don't save the settings, they revert back after a little while so you have a usable system.

I will set up a test rig and try a few options and report back. Of course maybe this is specific to the devices I am using:

HS1177 -> Micro MinimOSD -> Hawkeye 200Mw 5G8 VTX -> RX-LCD5802 (monitor with diversity)

ShikOfTheRa, 2016-01-12:
Actually it works different for me... I can swap between modes with no issues. Just tried now. I thought I recalled it needed a reboot to work properly but doesn't seem to today.

I think as you say - if your monitor / goggles don't auto detect either then that will be an issue.

ShikOfTheRa, 2016-02-17:
Added improvements for this into the roadmap for future.

closed 2016-02-17 by ShikOfTheRa