ShikOfTheRa/scarab-osd #174: Firmware update via Cleanflight

btimby, 2016-01-11:
Hi,

I just completed a pull request for Cleanflight that allows serial passthrough to a connected serial device such as an OSD. This is currently useful for changing the configuration.

https://github.com/cleanflight/cleanflight/pull/1620

I would love to be able to flash MWOSD releases using this method.

I am not overly familiar with how a minimosd is flashed other than that the DTR pin is used to reset the device, I am assuming that some bootloader code is invoked allowing the flashing process to proceed. I have also read that using the reset button allows flashing as well.

The micro minimosd does not have a reset button and the DTR pin is not connected to a flight control board. Thus my question is whether it is possible to implement a soft reset feature in MWOSD and further how this could be used to allow the Arduino tool to flash the OSD. My thought is that one could simply echo the soft reset command to the serial port and then execute avrdude. Perhaps this could also lead to firmware flashing from within the MWOSD GUI.

If you can provide me some hints I am happy to try and implement this feature and document it's use. I know this would be very useful to me (and probably others).

btimby, 2016-01-11:
Actually, I found this:

https://github.com/ShikOfTheRa/scarab-osd/blob/master/MW_OSD/GlobalVariables.h#L670

And this:

https://github.com/ShikOfTheRa/scarab-osd/blob/master/MW_OSD/MW_OSD.ino#L634

So apparently the above is possible if one could synchronize sending command 5 and running avrdude.

CrazyCoder, 2016-01-11:
That would be great for the tiny builds like Atom, especially when there is a PDB with integrated OSD. Thanks for the cleanflight patch and appreciate your efforts in this direction.

I think it should be possible to make some batch file that will send command 5 via COM port for reset and then call avrdude to flash the hex.

Similar approach is used to flash MotoLab Tornado FC. They send "R" command via serial and then call `STMFlashLoader.exe` to flash the firmware hex: [flash.bat](https://github.com/MotoLab/Cleanflight-MotoLab/blob/motolab/support/flash.bat).

ShikOfTheRa, 2016-01-11:
Thats a great feature to be able to program from GUI in situ. Frequently asked for.

For updating the OSD - probably a bit more complex. Timing has to be right too.
CF or MWOSD GUI issues the reboot OSD command when it detects a programming request. I am not 100% sure if this is same reboot 
Also CF will have to change baud rate on the fly to that required to program. Think its 57.6k from memory.

I think its achievable, but I think also a real pain to do. Good luck! Will help in any way I can. One day may well put a link in the GUI to edit / program from GUI.

btimby, 2016-01-11:
The reboot is a `jmp 0`, which jumps to the bootloader. Good chance it will go through the firmware upload process.

Timing is important since the bootloader only waits a short while for an upload to begin.

I will work on a PoC.

@ShikOfTheRa What are your long-term plans for the GUI? I ask because I found a lot of bits necessary for updating firmware within a Chrome App. I have long had the thought that Cleanflight might benefit from plugins (such as OSD control). With serial passthrough and the planned return from passthrough a plugin for MWOSD inside the Cleanflight UI should be possible. Being able to update firmware and change a few settings initially would be pretty killer.

tracernz, 2016-01-11:
@btimby A separate Chrome app would be much better. MWOSD supports a variety of non-CF firmwares and I think moving to support Cleanflight only would result in a fork.

btimby, 2016-01-11:
@tracernz certainly that is not my intention. I did not explain this well.

If MWOSD was configured via a Chrome app. Then Cleanflight or other Chrome Apps could EMBED it as a plugin. In addition to being it's own standalone configurator.

btimby, 2016-01-11:
Imagine that Cleanflight had an OSD menu item, which when clicked would send an embed request to the MWOSD configurator via the Chrome appview API. Cleanflight would set up passthrough and send connection info to MWOSD. The UI would be rendered in the right hand panel. MWOSD would be it's own standalone Chrome app that could establish it's own connection etc. when not used as a plugin.

tracernz, 2016-01-11:
@btimby That sounds reasonable. :+1:. Chrome does allow you to share resources between apps so it could be interesting to explore that.

ShikOfTheRa, 2016-01-12:
I can say I hate processing and the current GUI. 
That said - no plans to spend a lot of time building it in chrome to provide the same functionality it has today. It will take too long for little benefit to me. 
In hindsight I wish I had built it and the OSD from scratch as it would have taken a lot less time to develop and be much more efficient. it started as a few tweaks and just kept on growing.

error414, 2016-01-12:
I hate processing GUI too :(. On osx doesn't work properly (LSOpenURLsWithRole ....) , I have to use version for linux32 but it takes about 115% cpu load. I think that create GUI in chrome app extension 
can't take much time. 

BTW: thank you very much for this "update", it is very useful for me.

ShikOfTheRa, 2016-01-20:
You are welcome to create it if it doesn't take much time!!

error414, 2016-02-06:
Hi :), here you have demo of cores for strips editor, next step will be wysiwyg editor for screen. It is write in javascrit and canvas. http://error414.com/max7456/ Laste step will be form element and communications over serial.

error414, 2016-02-11:
core for screen editor is done, source code is here https://github.com/error414/max7456-GUI I am still working on it.

ShikOfTheRa, 2016-02-12:
Awesome - trying to get this running in chrome. got some permissions errors so will have to spend some time next week. Been wanting to have a look at the chrome stuff for some time. Thanks!

error414, 2016-02-12:
Thanks, you dont need run it as chrome app, they are just standalone html pages. In my chrome all works properly "Version 48.0.2564.109 (64-bit)" osx 10.11.1 (15B42). Here are live demos http://error414.com/max7456/edit_resolution.html , http://error414.com/max7456/edit_resolution.html , I am not JS programmer neither graphic designer but I do my best ;)

ShikOfTheRa, 2016-02-15:
Looks good! Actually I am not a programmer either... just a "security guard" . No programming experience... but I am determined and can read / learn...

error414, 2016-02-15:
my email is error414@error414.com , I think that it is not good place for discussion about it :).

closed 2016-02-16 by ShikOfTheRa