ShikOfTheRa/scarab-osd #175: Forced crosshair position is off center because of the new defaults

CrazyCoder, 2016-01-11:
It looks like https://github.com/ShikOfTheRa/scarab-osd/commit/07d03bec0c1260de5e23b923ee8cd51517057d37 has broken the forced crosshair display position.

You forgot to change the code inside https://github.com/ShikOfTheRa/scarab-osd/blob/master/MW_OSD/Screen.ino#L1794.

Should looks like this now:

```
  uint16_t position = getPosition(horizonPosition);
  screen[position-1] = SYM_AH_CENTER_LINE;
  screen[position+1] = SYM_AH_CENTER_LINE_RIGHT;
  screen[position] =   SYM_AH_CENTER;
```

ShikOfTheRa, 2016-01-11:
OK - updated in repository

ShikOfTheRa, 2016-01-11:
And thanks for guidance - I don't use this - so completely missed.

CrazyCoder, 2016-01-11:
Thanks for the quick fix!

closed 2016-01-11 by CrazyCoder

ghostface, 2017-11-25:
I have that code in my build but somehow the crosshair is still not centered?
I first thought it might be related to my widescreen c800t camera but its the same on my hs1177.
Cam is running on NTSC so there should be no PAL cropping which could affect it.

Also I saw this issue https://github.com/ShikOfTheRa/scarab-osd/issues/318 and the screencaps in there show the crosshair exactly as off-centered as I have it.

I doctored an image real quick to show what I mean:

[image]

Any advise?

ShikOfTheRa, 2017-11-29:
Note that many screen displays do not accurately presnet the full PAl or NTSC size so can appear off center. They are particularly variable with the height. Less common is width
You can just move them around with GUI to compensate.

ghostface, 2017-11-29:
Thanks for the reply! At first I kinda figured the horizon would be sort of auto centered but yeah manually moving it is kinda obvious now that I think about it :)

ghostface, 2019-04-10:
I know this is super old but I am now using 1.8 on a silverware whoop build and I no longer can move the the horizon item around in the configurator to center it. Any idea/setting I am missing?