ShikOfTheRa/scarab-osd #177: Camera OSD passthrough

zscally, 2016-01-25:
Not an Issue but wondering if a pass through would be possible? my limited understand of circuits and C code limits me from doing this but I am a programmer and have a small understanding of whats going on. To my self here is the scenario:

1) An option in the WMOSD menu drops the chip into CAMERA OSD mode.
    a. this disables the WMOSD interface and try's to send a signal to the OSD pin on the camera relaying stick input from PWM D11 to an non used PWM D09 / D10?  interface that the Camera is connected to. 
   b. Would require a wire to be going from the camera to the PWM port on the MINIMOSD.

2) Once configuration has been made to camera "exit" would be selected from the cameras OSD menu causing the camera to reset "is this a measurable reset"? if so this would cause the MINIMOSD To go back into WMOSD mode.

I know I don't have any insight into this and I'm speculating a lot here but just wondering if its possible.

HS1177 camera uses one wires to send a PWM impulse to the camera to cause it to fall into a configuration mode I'm sure most cameras work the same.

[image]

tracernz, 2016-01-25:
Alternately you can setup an OSD switch on your flight controller to disable the OSD from your radio.

zscally, 2016-01-25:
Yeah you're right that functionality is already there! -> but now we need to be able to pass stick input from the flight controller to the ATMEGA328 to translate into something that the camera OSD port can understand

tracernz, 2016-01-25:
Ah, that's a nice idea!

zscally, 2016-01-25:
Even running something like an Alien 5" where the camera is relatively easy to get to the camera is not. Currently having to break down the front end of the quad to beable to plug in my hacked cord to program the camera. 

Typically you set your settings one and your done. But for more experienced pilots _nut jobs_ like me flying in exotic places is always on the rise example: parking garage, Cavern etc where light changes from place to place and having optimal viewing experience is a must!!!

https://www.youtube.com/watch?v=TkH1Ia298sY

ShikOfTheRa, 2016-01-26:
No plans to do this unfortunately - just too many variations - and many of the cameras I have, they are different - some send different voltages (usually single wire ones), some send a signal (unknown and custom format) and some use multiple wires.

ShikOfTheRa, 2016-02-17:
Closing as no action planned

closed 2016-02-17 by ShikOfTheRa