ShikOfTheRa/scarab-osd #178: NOCONTROLLER doesn't work in version 1.5.2.3

19Gerhard85, 2016-01-28:
When selecting NOCONTROLLER in Config.h an error occurs:

```
Arduino: 1.6.7 (Windows 10), Board: "Arduino Pro or Pro Mini, ATmega328 (5V, 16 MHz)"

In file included from C:\scarab-osd-1.5.2.3\scarab-osd-1.5.2.3\MW_OSD\Serial.ino:86:0:

GlobalVariables.h:141: error: 'MENU1' was not declared in this scope

 int8_t configPage=MENU1;

                   ^

C:\scarab-osd-1.5.2.3\scarab-osd-1.5.2.3\MW_OSD\Serial.ino: In function 'void loop()':

Serial:409: error: 'MENU0' was not declared in this scope

         len++;

                    ^

C:\scarab-osd-1.5.2.3\scarab-osd-1.5.2.3\MW_OSD\Screen.ino: In function 'void displayConfigScreen()':

Screen:1196: error: 'MENU0' was not declared in this scope

   if(configPage==MENU0)

                  ^

Screen:1489: error: 'MAXPAGE' was not declared in this scope

     if(configPage > MAXPAGE)configPage=MINPAGE;

                     ^

C:\scarab-osd-1.5.2.3\scarab-osd-1.5.2.3\MW_OSD\Serial.ino: In function 'void handleRawRC()':

Serial:585: error: 'MENU0' was not declared in this scope

         if(configPage == MENU0) {

                          ^

Serial:614: error: 'MENU9' was not declared in this scope

  if(configPage == MENU9 && COL == 3) {

                   ^

C:\scarab-osd-1.5.2.3\scarab-osd-1.5.2.3\MW_OSD\Serial.ino: In function 'void serialMenuCommon()':

Serial:640: error: 'MAXPAGE' was not declared in this scope

     if(configPage<MINPAGE) configPage = MAXPAGE;

                                         ^

Serial:641: error: 'MAXPAGE' was not declared in this scope

     if(configPage>MAXPAGE) configPage = MINPAGE;

                   ^

C:\scarab-osd-1.5.2.3\scarab-osd-1.5.2.3\MW_OSD\Serial.ino: In function 'void configExit()':

Serial:864: error: 'MENU1' was not declared in this scope

   configPage=MENU1;

              ^

exit status 1
'MENU1' was not declared in this scope

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.
```

ShikOfTheRa, 2016-01-29:
Yes - my error. Thanks. Resolved and uploaded to repository

19Gerhard85, 2016-01-29:
Thanks will test it with the new release.

closed 2016-02-17 by ShikOfTheRa