ShikOfTheRa/scarab-osd #179: Display PIDs while flying

KevinVecchione, 2016-02-01:
It would be nice to get visual confirmation of PID for R/P/Y while tuning with Cleanflight adjustment switches during flight. Is something like this theoretically possible? I'm open to helping out and contributing this to the code if someone can give me a quick orientation of what's what in the source.

Thanks!

ShikOfTheRa, 2016-02-02:
This is definitely possible.
I can add to the wish list however, not recommended as a permanent option and low on priority list. Mainly because...
1 adds processing time for serial for both the FC and the OSD. Everything else will slow down.
2 lack of code space. if anyone implements, bust be able to be excluded with #define

ShikOfTheRa, 2016-02-17:
Have no short term plans to implement. Have added to the roadmap "wishlist" for someone to implement if they desire.

closed 2016-02-17 by ShikOfTheRa