ShikOfTheRa/scarab-osd #180: Support for KISS FC

gregorg, 2016-02-07:
The Kiss flight controller from Flyduino.net has his own firmware: http://ultraesc.de/downloads/KISS_OSD.zip

Kiss FC + Kiss 24A ESC adds sensors: temperature and current conumption per ESC. Also, global current consumption in mA is very useful.

Is there any plan to add it ? Or can you tell me the way to add it myself ?

ShikOfTheRa, 2016-02-08:
This is quite cool. Not seen before.

Unfortunately couldn't be added if using with a FC like a naze. Is that what you mean?

gregorg, 2016-02-08:
Nope. Kiss FC is a flight-controller, with proprietary firmware. It's supports telemetry and add those sensors.

ShikOfTheRa, 2016-02-09:
OK I wasn't aware they had an FC Assumed it came from serial bus to ESC or soemthing. Even cooler !!

Had a look at their web page. Seems it's closed source so no info. Had a quick look at the OSD code and it is not using MSP so would have to reverse eng.

As a result - too much time so sorry no plans from me... 
Would be happy for someone to add in support subject to being able to exclude additional code via #defines. I think it's not to difficult...

ShikOfTheRa, 2016-02-17:
gregorg - if the kiss guys would like to add this support into MWOSD can you get them to contact me.

gregorg, 2016-02-18:
I'm in contact with Felix from Flyduino, he will transmit me the kiss serial protocol.

Please re-open this issue ;)

gregorg, 2016-02-19:
the serial connection is 8bit 115200baud full duplex you can get the datas at serial1 (RX/TX pins next to the USB port) the FC will only answer to requests of on the serial, as long as there is no USB plugged in.

to request the telemetry data simply write 0x20 to it. the replay function lookes like this:

	// GUI viewable variables
	
	serial_send_byte(5); // startByte
	serial_send_byte(154); // BytesNum the byte count that will be sent
	
	serial_send_floats(RXcommands,8); // 16
	serial_send_byte(Armed); // 17
	serial_send_word((uint16_t)(LiPoVolt*100)); // 19
	serial_send_words(GyroXYZ,3); //25
	serial_send_words(ACCXYZ,3); // 31

	serial_send_word((int16_t)(angle[0]*100)); // 33
	serial_send_word((int16_t)(angle[1]*100)); // 35
	serial_send_word((int16_t)(angle[2]*100)); // 37
	
	serial_send_word(I2C_Errors); // 39
	serial_send_word(calibGyroDone); //41
	serial_send_byte(failsave); //42
	serial_send_word((uint16_t)debug*1000); //44
	serial_send_byte(0); //45
	
	serial_send_floats(GyroRaw,3); //51
	serial_send_floats(ACCRaw,3); //57
	
	serial_send_floats(ACCtrim,2); // 61
	serial_send_floats(ACCAng,2); // 65
	serial_send_byte(mode); // 66
	serial_send_word((uint16_t)debug1*1000); //68
	serial_send_words(PWMOutVals,6); // 80
	serial_send_word((uint16_t)debug2*1000); //82
	serial_send_byte(idleTime);//83
	
	serial_send_words(ESC_Telemetrie[0],5); // 93
	serial_send_words(ESC_Telemetrie[1],5); // 103
	serial_send_words(ESC_Telemetrie[2],5); // 113
	serial_send_words(ESC_Telemetrie[3],5); // 123
	serial_send_words(ESC_Telemetrie[4],5); // 133
	serial_send_words(ESC_Telemetrie[5],5); // 143
	serial_send_byte(MaxTemp); // 144
	serial_send_word(MinVolt); // 146
	serial_send_word(MaxAmps); // 148
	serial_send_word(usedMAH); // 150
	serial_send_word(MaxRPMs); // 152
	serial_send_word(MaxWatt); // 154
	
	
	serial_send_byte(checkSum); // endByte 

- the channel order is throttle, roll, pitch, yaw, aux 1234

- gyro and acc datas are send X,Y,Z//0,1,2

- the ESC's telemetry is for each ESC 

  0: temperature in 1°C, 
  1: voltage in 0.01V 
  2: current in 0.01A
  3: consumption in 1mah
  4: erpm in 100erpm

- the checksum becomes useless as there are too many bytes now. i will add a real CRC in future. if you still want to use it, it is the average of all bytes send (excluding the two start bytes and its self)

- floats are used *1000 so (float)1.0f == (int16_t)1000

- 16bit wariables are sent with high byte first

gregorg, 2016-02-23:
Can you re-open it ?

ShikOfTheRa, 2016-02-23:
If anyone picks this up... take not of 
  #define SERIALBUFFERSIZE 150 in serial.ino

It will need to be amended to suit - but also note will have to verify stack/heap memory is not blown

matolowa, 2016-02-25:
hi ,gregorg
If is  possible please write to kiss OSD stopwatch .i use it in reduce mode.Bottom middle would be nice. :) thank you

gregorg, 2016-02-25:
@matolowa you mean on original Kiss OSD firmware ?

matolowa, 2016-02-25:
yes,or is it impossible?

gregorg, 2016-02-25:
it is, but it's not the purpose of this issue ;) I've created a github repository to enhance the original Kiss OSD here: https://github.com/gregorg/KISS_OSD
Please create a new issue on this repository.

ShikOfTheRa, 2016-04-11:
@gregorg - If  prepare the MWOSD to theoretically work with KISS OSD would you be interested in testing and fine tuning any variable mapping. 
I'll get it to the point where its probably 90%

matolowa, 2016-04-11:
greg i´m interested to testing you work.your kiss osd everyone used in
Slovakia :),,everyone is satisfied ..nickname is the best :) If we fly
five..

gregorg, 2016-05-02:
@ShikOfTheRa I'm sorry I forgot to reply to your proposal ... Sure, I'm OK to test it ! I'm glade to see that it is ready for testing !
I'll have a fresh kiss fc and micro mini osd to test it soon.

ShikOfTheRa, 2016-05-03:
Its not ready yet! Just completed mavlink ready for testing. Probably LTM next then KISS. I'll notify you when have something. Thanks for reply :)

matolowa, 2016-05-03:
ok.I will be waiting..

ShikOfTheRa, 2016-05-25:
Can anyone clarify:
For RC channels:
Guide above indicates RC data is sent as float..., but values are 16 bit rather than 32bit float so is this half float (unusual)
serial_send_floats(RXcommands,8); // 16

In the code, the data appears to contradict and indicate its 16 bit it with a range of prob -500 to +500??

I hope its the latter :)

gregorg, 2016-05-25:
If it helps you, take a look at the original Kiss OSD source code: https://github.com/gregorg/KISS_OSD/blob/master/src/KISS_OSD.ino

ShikOfTheRa, 2016-05-25:
Yeah in that code it appears that it is 16 bit int. with range of at least -250 to +250

Ah well I've done it like that and ignore the float method. Lets see what happens when someone tests!

ShikOfTheRa, 2016-05-25:
Also I think mode just determines full screen/minimal screen. 
Are there any different flight modes in KISS. Sorry for dumbass questions - just never seen / used !

Think I'll just do the basics and let someone else with kit to test  develop further.

gregorg, 2016-05-25:
there is 2 flight modes in Kiss: acro and horizon

ShikOfTheRa, 2016-05-25:
In the code, the mode indicator seems to change the screen displayed rather than indicate acro or horizon. Cant see anything for flight mode.

Welcome to try first cut.... I have to do the amperage as its currently using method used by mavlink...
Its not easy without knowledge or testing! Probably find anumber of items not quite correct!
https://github.com/ShikOfTheRa/scarab-osd/tree/SHIKI

Once get basics done, can add in individual ESC info etc.

llambkin, 2016-05-27:
Id love to test, but I have no coding skills, so I dont know if I would be of any help....

llambkin, 2016-05-31:
OK....I have thrown the program on the OSD, The voltage appears to work ok, as does the amperage counter. However, when I have finished a flight it shows the statistics for a split second only.

Quick question. Do you think that KISS would allow for the changing of PIDs in the OSD? Probably asking a bit much for a closed source FC.

fedorcomander, 2016-05-31:
Its possible to change pids. No problem here. Protocol is rather simple and you can see it in the GUI projects on the github :) But u cannot save it while LIPO is connected (yet)...  And you need something to power vtx and osd.... 

And one more thing. float data being send as integer x1000 :)))

ShikOfTheRa, 2016-06-01:
@fedorcomander thanks for float info. I understand now. No float being sent....
It was confusing me because it was saying float....  !!

@llambkin - thanks to fedor post above I think the throttle info would be to high making the end of flight flash off/on 

I think ...
throttle channel  is being send as 1000 + (0-1000)
all other 1500 + (-500 to +500)

Also I changed the Amperage calculation to be direct from ESC. Be interested to hear if this still works and is accurate.

Also, can you see if access the OSD menu?

test code here...
https://github.com/ShikOfTheRa/scarab-osd/tree/SHIKI

fedorcomander, 2016-06-01:
obj.RXcommands[0] = 1000 + ((data.getInt16(i + 0, 0) / 1000) * 1000);
			obj.RXcommands[1] = 1500 + ((data.getInt16(i + 2, 0) / 1000) * 500);
			obj.RXcommands[2] = 1500 + ((data.getInt16(i + 4, 0) / 1000) * 500);
			obj.RXcommands[3] = 1500 + ((data.getInt16(i + 6, 0) / 1000) * 500);
			obj.RXcommands[4] = 1500 + ((data.getInt16(i + 8, 0) / 1000) * 500);
			obj.RXcommands[5] = 1500 + ((data.getInt16(i + 10, 0) / 1000) * 500);
			obj.RXcommands[6] = 1500 + ((data.getInt16(i + 12, 0) / 1000) * 500);
			obj.RXcommands[7] = 1500 + ((data.getInt16(i + 14, 0) / 1000) * 500);
			obj.Armed = data.getUint8(i + 16);
			obj.LiPoVolt = data.getInt16(i + 17, 0) / 1000;
			obj.GyroXYZ[0] = data.getInt16(i + 19, 0);
			obj.GyroXYZ[1] = data.getInt16(i + 21, 0);
			obj.GyroXYZ[2] = data.getInt16(i + 23, 0);
			obj.ACCXYZ[0] = data.getInt16(i + 25, 0);
			obj.ACCXYZ[1] = data.getInt16(i + 27, 0);
			obj.ACCXYZ[2] = data.getInt16(i + 29, 0);
			obj.angle[0] = data.getInt16(i + 31, 0) / 1000;
			obj.angle[1] = data.getInt16(i + 33, 0) / 1000;
			obj.angle[2] = data.getInt16(i + 35, 0) / 1000;
			obj.I2C_Errors = data.getInt16(i + 37, 0);
			obj.calibGyroDone = data.getInt16(i + 39, 0);
			obj.failsave = data.getUint8(i + 41);
			obj.debug = data.getUint16(i + 42, 0) / 1000;
			obj.foundRX = data.getUint8(i + 44);

			obj.GyroRaw[0] = data.getInt16(i + 45, 0) / 1000;
			obj.GyroRaw[1] = data.getInt16(i + 47, 0) / 1000;
			obj.GyroRaw[2] = data.getInt16(i + 49, 0) / 1000;
			obj.ACCRaw[0] = data.getInt16(i + 51, 0) / 1000;
			obj.ACCRaw[1] = data.getInt16(i + 53, 0) / 1000;
			obj.ACCRaw[2] = data.getInt16(i + 55, 0) / 1000;
			obj.ACCtrim[0] = data.getInt16(i + 57, 0) / 1000;
			obj.ACCtrim[1] = data.getInt16(i + 59, 0) / 1000;
			obj.ACCAng[0] = data.getInt16(i + 61, 0) / 1000;
			obj.ACCAng[1] = data.getInt16(i + 63, 0) / 1000;
			obj.mode = data.getUint8(i + 65);
			obj.debug = data.getUint16(i + 66, 0) / 1000;
			obj.PWMOutVals[0] = data.getInt16(i + 68, 0);
			obj.PWMOutVals[1] = data.getInt16(i + 70, 0);
			obj.PWMOutVals[2] = data.getInt16(i + 72, 0);
			obj.PWMOutVals[3] = data.getInt16(i + 74, 0);
			obj.PWMOutVals[4] = data.getInt16(i + 76, 0);
			obj.PWMOutVals[5] = data.getInt16(i + 78, 0);
			obj.debug2 = data.getUint16(i + 80, 0) / 1000;
			obj.idleTime = data.getUint8(i + 82);

			obj.ESC_Telemetrie0[0] = data.getInt16(i + 83, 0);
			obj.ESC_Telemetrie0[1] = data.getInt16(i + 85, 0);
			obj.ESC_Telemetrie0[2] = data.getInt16(i + 87, 0);
			obj.ESC_Telemetrie0[3] = data.getInt16(i + 89, 0);
			obj.ESC_Telemetrie0[4] = data.getInt16(i + 91, 0);

			obj.ESC_Telemetrie1[0] = data.getInt16(i + 93, 0);
			obj.ESC_Telemetrie1[1] = data.getInt16(i + 95, 0);
			obj.ESC_Telemetrie1[2] = data.getInt16(i + 97, 0);
			obj.ESC_Telemetrie1[3] = data.getInt16(i + 99, 0);
			obj.ESC_Telemetrie1[4] = data.getInt16(i + 101, 0);

			obj.ESC_Telemetrie2[0] = data.getInt16(i + 103, 0);
			obj.ESC_Telemetrie2[1] = data.getInt16(i + 105, 0);
			obj.ESC_Telemetrie2[2] = data.getInt16(i + 107, 0);
			obj.ESC_Telemetrie2[3] = data.getInt16(i + 109, 0);
			obj.ESC_Telemetrie2[4] = data.getInt16(i + 111, 0);

			obj.ESC_Telemetrie3[0] = data.getInt16(i + 113, 0);
			obj.ESC_Telemetrie3[1] = data.getInt16(i + 115, 0);
			obj.ESC_Telemetrie3[2] = data.getInt16(i + 117, 0);
			obj.ESC_Telemetrie3[3] = data.getInt16(i + 119, 0);
			obj.ESC_Telemetrie3[4] = data.getInt16(i + 121, 0);

			obj.ESC_Telemetrie4[0] = data.getInt16(i + 123, 0);
			obj.ESC_Telemetrie4[1] = data.getInt16(i + 125, 0);
			obj.ESC_Telemetrie4[2] = data.getInt16(i + 127, 0);
			obj.ESC_Telemetrie4[3] = data.getInt16(i + 129, 0);
			obj.ESC_Telemetrie4[4] = data.getInt16(i + 131, 0);

			obj.ESC_Telemetrie5[0] = data.getInt16(i + 133, 0);
			obj.ESC_Telemetrie5[1] = data.getInt16(i + 135, 0);
			obj.ESC_Telemetrie5[2] = data.getInt16(i + 137, 0);
			obj.ESC_Telemetrie5[3] = data.getInt16(i + 139, 0);
			obj.ESC_Telemetrie5[4] = data.getInt16(i + 141, 0);

			obj.ESC_TelemetrieStats[0] = data.getInt16(i + 142, 0);
			obj.ESC_TelemetrieStats[1] = data.getInt16(i + 144, 0);
			obj.ESC_TelemetrieStats[2] = data.getInt16(i + 146, 0);
			obj.ESC_TelemetrieStats[3] = data.getInt16(i + 148, 0);
			obj.ESC_TelemetrieStats[4] = data.getInt16(i + 150, 0);
			obj.ESC_TelemetrieStats[5] = data.getInt16(i + 152, 0);

fedorcomander, 2016-06-01:
Hope this help :)

fedorcomander, 2016-06-01:
all this code available in https://github.com/fedorcomander/kissfc-chrome-gui

ShikOfTheRa, 2016-06-01:
Yes - much thanks. I just saw your link above. Very helpfull

llambkin, 2016-06-01:
Hello Guys,

I have done an actual flight with the test code from a few days ago. You are right Shik, it was flashing on/off a lot thoughout the flight. Also, yes I can access the menu, it is far more responsive than Naze F3 targets, that was maybe because the PWM was maxed a lot sooner.

I will throw the new code on there tonight and give it a test, I will report back here.

Here is a video of the flight in question.

llambkin, 2016-06-01:
++ Thanks for the help fedorcomander. Lets hope that they fix the save feature so it allows for changing PIDs via OSD. Or is it you that is working on this?

llambkin, 2016-06-02:
Ok....Ive tested the current build Shik. The amperage appears to work now. But the amperage counter doesnt work at all. It sat on 0 the whole time.

Also, I had a reset during flight. It said disarmed, then armed and continued as normal.

Flight video to follow.

ShikOfTheRa, 2016-06-02:
Thanks for vid - VERY usefull.
MaH fixed - my error.
I think I know cause of the armed reset. Probably accounts for other glithes - including the High max amps on summary.

So throttle and summary screen looks OK - does the menu work OK - get into it OK, change setting and exit?  

Try to get last bits done tomorrow

gregorg, 2016-06-02:
Awesome ! I do not have a testing frame today, I will have one in couple weeks and will test it for sure :)

Another usefull features to add is ESC temperatures, and maybe caliGyroDone ?

mart1n, 2016-06-03:
Oh man! I'm excited to see that support for the KISS FC is being added to MWOSD. I've been having issues with the KISS OSD firmware. The entire video feed from my vtx is full of random characters, mostly the letter 'V', from top to bottom. I never loaded the KISS fonts file though, so that might be it. But I'm not going to worry about trying to get it to work and I'm going to move straight over to this branch of MWOSD. I'll probably even help with some development.

llambkin, 2016-06-03:
Yes Shik, I am able to enter the menu, and change settings and exit :smile: 

Thanks heaps for your coding ninja skills. Will test the new code as soon as possible. I am guessing it would take a fair bit of work (changing the configurator) to add the code for individual ESC amps/temps/RPM?

ShikOfTheRa, 2016-06-03:
Its now merged into the Master as its getting closer to being usable.. had a couple of issues with sync so hopefully OK

https://github.com/ShikOfTheRa/scarab-osd/

what needs testing (I think!)
Does it still work (have added the checksum to validate data coming through - if wrong will see nothing !)
Amps - is it accurate and smoother
Horizon bar - does it move at all / roughly track horizon / ?
Has in flight screen quick disarm glitching stopped.

@llambkin thanks. Less ninja, more turtle.... 
I want to get the basics in that MWOSD supports today. Someone can then look to add

@mart1n  - welcome on board. Happy to take any improvements

@gregorg  - wanted to get the basics in place so others can expand. It should be easy to implement though I'll probably not be adding those bits myself (not exactly easy when not having a frame with kiss controller to test! - but happy to take any submissions)

llambkin, 2016-06-03:
@ShikOfTheRa, I will test all of the above. I don't really fly with the horizon on, but will test for those who do.

Will report back tomorrow with another video.

mart1n, 2016-06-03:
@ShikOfTheRa - In my previous comment I mentioned that the KISS OSD firmware filled my vtx video feed with characters from top to bottom. After flashing MWOSD, the same issue exists!

If you look at the top of the KISS_OSD.ino file in the KISS OSD FW you notice the second group of macros reads...

`// MAX7456 Charset (change if you get sensless signs)
//=============================
#define USE_MAX7456_ASCII
//#define USE_MAX7456_MAXIM
`

I had to change this from USE_MAX7456_ASCII to USE_MAX7456_MAXIM. In the KISS OSD FW to fix my issue. The USE_MAX7456_* macros are used to determine the character encoding used.

I noticed the MAX7456 APIs that the KISS OSD FW uses is different from what MWOSD is using.

Does MWOSD already have a method for switching character encoding? I briefly looked for a way to switch MWOSD from MAX7456_ASCII to MAX7456_MAXIM and didn't see anything.

I will mention in my search through rcgroups.com I have found 2 other people that are having this exact issue with MWOSD firmware. It's not an issue with font loading either as I tried reloading fonts with USB power as well as with Lipo power.

Thanks!

Martin

mart1n, 2016-06-03:
Here is an example of what my vtx video feed looks like....
[http://static.rcgroups.net/forums/attachments/5/7/6/3/3/6/a8858368-194-osderror.jpg](http://static.rcgroups.net/forums/attachments/5/7/6/3/3/6/a8858368-194-osderror.jpg)

llambkin, 2016-06-04:
OK, I have tested the code.

It appears that the amps is working and correct.
The horizon appears to work, but is back to front.
I didnt have a long enough flight to test if the in flight glitching has stopped.

I will leave the code on there and take it for a proper test later.

Regards -Liam

llambkin, 2016-06-05:
I have done a longer flight today.
The in flight resets are still happening, I was looking for how many amps I have used, and it only said 70mah..... Then I noticed it reset to 0, and it did this several times through the flight.

Sorry I dont have a video of this happening. I will get one tonight when I go out for an event.

llambkin, 2016-06-06:
Hahahahaha, I just realised, the horizon isnt back to front.
So, that just leaves the reset glitch.

ShikOfTheRa, 2016-06-07:
OK I thought adding checksum for the data would remove the "reset"  or at least minimise it.
What is interesting - the mah counter comes from the FC, so if that is resetting when you get the OSD glitch, then its likely to be that your FC is resetting that info.... Look forward to next vid. Sorry for delay - been away for  few days

llambkin, 2016-06-07:
I will make a video tonight, I cant get it to do it in the garage.....

llambkin, 2016-06-08:
Video as promised.....A little late, sorry :)
https://youtu.be/9AiNuXj47Rs

ShikOfTheRa, 2016-06-08:
No worry... Perfect! Actually I think the resets are solved/improved - I didn't see any... 
The mah counter is a separate issue. I can see the problem in my code. I'll simulate test and advise back...
The frequent low vlts are annoying - i will try to put some code to improve that in the future. For now can comment this part out...
//#define DISP_LOW_VOLTS_WARNING      // Alerts with additional text warning if low voltage

ShikOfTheRa, 2016-06-08:
OK, that done tested and uploaded.
I think only time will tell now to see if we get any more screen resets.
Thanks for your time to do the testing and uploads. I know its a pain in the butt....

I think we have the basic in place for KISS MW-OSD :) something we can build and add to.
Keep us posted if see any more issues.

llambkin, 2016-06-08:
Awesome. I will test and report back. Its great to collaborate with nice people :D
Yeah, I have some saggy LiPo's and KISS FC spits out the data so quickly that both my TX (Taranis) and OSD bitch about voltage all the time. I think its about time to get some better LiPo's.

 Some code to put a delay of about 5 secs of low voltage would be nice. It rarely gets hammered for more than 5 seconds at a time. I have put a logical switch into my TX that triggers an announcement only after 5 seconds of sag below 10.5v

Thanks again for your skills. I am more than happy to test the code out since it is great benefit to myself and others once we get it right. Is it a big job to code the individual ESC information into the GUI and OSD, is that something that someone else is likely to have to do?

llambkin, 2016-06-09:
I have tested the current build with my copter. It all works as expected, no resets, and I can now rely on it for mAh used :D

llambkin, 2016-06-09:
https://youtu.be/nPt2Xlb3CXg

ShikOfTheRa, 2016-06-10:
Awesome!! Thanks to you guys testing and nagging me I think we have at least the essentials working now. 
For adding indivdual ESC data its not difficult. Just some time. Figuring out best way to display really.

How usefull is that ESC info? Think I would prefer to have PIDS adjustable in menu over that

llambkin, 2016-06-10:
All I would want is Temp and Amps of ESC. The current way of displaying the info is to put the info in each corner according to the layout of the motor order.

Yeah, adjusting PID via OSD would be great, but I don't know when Flyduino plan on making that available to us. They currently have a problem where if you send a save command to the FC, it flips out. So their fix for that is not being able to save with a lipo connected.

So i figured that someone could work on the features available to us right now.

ShikOfTheRa, 2016-06-10:
OK. Ill have a think. I'd hate to fly with that info on all the time - so prob make it one of the switchable layout options

ShikOfTheRa, 2016-06-10:
@mart1n 

Maybe the dev latest version on MWOSD will be of use to you no llabkin confirmed its looking OK. 

Alternatively, in mwosd, can flash fonts without the gui using this method:

/*--------------------------       INITIALISATION options       ----------------------------------------------------*/
// Ignore this section unless you know you need to use it !!
// This section contains initialisation options that only require to be run once.
// Once the initialisation has completed, all sections should be commented and the sketch re-uploaded.
// Font upload will take 90 seconds after upload is completed. If connected to a camera, you will see teh font table displayed.

//#define LOADFONT_DEFAULT         // Uncomment to force an upload of default font instead of using GUI

llambkin, 2016-06-10:
I put the info to the back of my mind, I totally ignore it.I flew an entire pack without looking at the amperage today. I mainly want it for troubleshooting, its to look back on it after a flight, if one motor is a bit stuffed, then the ESC will draw substantially more current, and the ESC will get hotter.

But you are right, PIDs would be more important. Maybe you could start on just getting them to display, that would serve the purpose of having video evidence of your PIDs before you fly. Also we could do some testing on adjusting PIDs, since I can plug in a USB power pack to the FC, since I can save values when the FC is powered off USB. Also the FC powers the RX too.....but not the VTX.....hmmmmm

ShikOfTheRa, 2016-06-10:
Hopefully have some KISS stuff on the way soon to build a test rig. Once get that can have a look at the extras unless someone wants to jump in and do them.

llambkin, 2016-06-10:
I will always be available to test :D

[24 more replies]

closed 2020-11-11 by ShikOfTheRa