ShikOfTheRa/scarab-osd #184: Acro + Indicator

zobi, 2016-02-24:
Hello, is it possible to add an acro + indicator in OSD ?
Thanks a lot

ShikOfTheRa, 2016-02-25:
I am not aware of acro+, can you provide a link to it describing it - or ideally code where it is?

zobi, 2016-02-25:
What is Acro + :
https://github.com/borisbstyle/betaflight/wiki/Frequently-Asked-Questions#what-is-acro-plus-

Normaly i think it's coded in : https://github.com/borisbstyle/betaflight/blob/betaflight/src/main/flight/mixer.c

;)

ShikOfTheRa, 2016-02-25:
OK, Yes - it is possible. 
I think for ICON - instead of ACRO - will display AC+
I'll put it onto development roadmap

ShikOfTheRa, 2016-02-25:
OK - maybe it is not possible...
I will have to check this in more detail. I am not sure if Arduino compiler can handle the higher number used to represent acro plus mode.

ShikOfTheRa, 2016-02-25:
OK - this is added in for the next release. Hopefully next week if no bugs found
You may test it by accessing the beta test area on mwosd.com

dustin, 2016-02-25:
This sounds different from the Acro+ I'm used to.  But this is what mine looks like:  http://gfycat.com/MixedMagnificentAdouri

The downside is, AFAICT, there's not an actual mode indicator sent along.

ShikOfTheRa, 2016-02-26:
There is an acro+ indicator sent with betaflight. I have no idea if the same as not tried betaflight yet, but 
in serial_msp we have this....
    { BOXACROPLUS, "ACRO PLUS;", 29 },
Its not there in cleanflight, baseflight or multiwii....

closed 2016-03-01 by ShikOfTheRa