ShikOfTheRa/scarab-osd #187: Sending VBat to XRacer F303

SK1701, 2016-02-29:
The new [XRacer F303](http://www.fpvmodel.com/x-racer-f303-flight-controller_g1106.html) seems like an excellent flight controller to run Cleanflight/ Betaflight but it lacks one important feature - VBat monitoring. I was wondering if it would be possible to send VBat data from the BAT pads on a MinimOSD over the serial port to the FC to allow use of features such as a buzzer and PID scaling with cell voltage? Kind of the reverse of the normal process of sending VBat from FC to the OSD..

ShikOfTheRa, 2016-02-29:
Yes it is possible - but would require coding for CF/BF etc to be in place first. I don't believe it is there today.

ShikOfTheRa, 2016-03-07:
I'm going to close this as not possible to implement without this functionality existing in CF/BF.
I'm not sure its such a good idea to rely upon this for pid scaling. It would scare me!

closed 2016-03-07 by ShikOfTheRa