ShikOfTheRa/scarab-osd #192: Strange behaviour with HB-Elevation enabled

amoksworld, 2016-03-06:
Hi,

when HB-Elevation is active, the additional lines change their distance to the horizon line in dependency of the roll angle.

If you roll to the right (while pitch angle keeps 0°), the top line comes closer to the horizon line (as the angle increases to 90° they hit each other) and if you roll left, the bottom line comes closer to the horizon line in the same way.
The strange thing is, that only one line moves as a function of roll angle.

I would prefer if there were some more HB-Elevation lines (like in a real HUD) so that they can act as an pitch indicator (perhaps every 10° or like in real a short one for 5° and a longer one for 10° steps).

I saw this in 1.5 and 1.6 beta.

ShikOfTheRa, 2016-03-07:
confirmed bug - only occurs if use this... I guess noty many do as its been missed in all testing
#define FULLAHI                   // Enable to display a slightly longer AHI line

Will investigate and provide an update

ShikOfTheRa, 2016-03-07:
OK - this is updated with a resolution in main repository now.

Thanks for pointing out

closed 2016-03-07 by ShikOfTheRa

amoksworld, 2016-03-07:
Thank you very much, I will check this out!