ShikOfTheRa/scarab-osd #193: TBS TBS Powercube Current via i2c

DaTwo, 2016-03-07:
Currently the TBS Powercube requires one to purchase the rather large TBS OSD in order to get the current displayed. This information is output from a connector on the TBS Powercube labeled BST.  The port as far as I can tell is this...

left to right

1  3.3v
2  BATT +
3  GND
4  SCL
5  SDA

I should be able to provide a dump and do some testing on this. Any chance you would be interested in implementing this into MWOSD? This would be of great use to those of us that are using this on the mini quads.

Thanks

tracernz, 2016-03-07:
I thought Cleanflight supports bst? There was a PR at one point.

DaTwo, 2016-03-07:
Any idea how to enable it? I am gonna see if I can locate any info. Would be great if its already there.

tracernz, 2016-03-07:
Ah, seems they don't want it: https://github.com/cleanflight/cleanflight/pull/1657

tracernz, 2016-03-07:
Looks like it's in betaflight... and is just MSP over I2C? https://github.com/borisbstyle/betaflight/blob/betaflight/src/main/io/i2c_bst.c

DaTwo, 2016-03-07:
OK, I am running the latest betaflight but when I turn on the current meter it appears to be tied to something else such as another voltage from an adc (in cleanflight config it stays basically a constant even while spinning up the motors).  Any idea on how to make this work? Just want mah used to be on the MWOSD.

DaTwo, 2016-03-07:
I am getting this under tasks in CLI so BST code must be active...

14 - (BST_MASTER_WRITE) max: 6us, avg: 0us, rate: 99hz, total: 217ms
15 - (BST_MASTER_PROCESS) max: 6us, avg: 0us, rate: 49hz, total: 74ms

ShikOfTheRa, 2016-03-07:
I2c pins are not available on mini osd so will not be progressing this. Sorry

DaTwo, 2016-03-07:
Looks like where the fix needs to be is in betaflight. From what I can tell BST (Black Sheep Telemetry) is implemented partially in betaflight. Since there is no input on the FC for current the standard ADC way maybe the current sensor can be implemented in the  "COLIBRI_RACE" target. It is connected via i2c port 2. Thanks anyways for your help.

ShikOfTheRa, 2016-03-07:
OK - thanks all - I'll close this

closed 2016-03-07 by ShikOfTheRa