ShikOfTheRa/scarab-osd #194: Support for Kylin 250 FPV PDB (All-In-One Board)

amoksworld, 2016-03-07:
On the Kylin 250 FPV (http://www.kylin250.com/) there is a single "mainboard" with an CC3D and minimOSD OnBoard which are hard wired (over the board layout) with the Flexy-Port of the CC3D. There is a voltage meter connected to analouge pin 6 of the minimOSD based on a voltage divider, too.
But in config.h it is only possible to select "SWAPVOLTAGEPINS" (what cahnges from A0 to A2).
So you have to go to Def.h and set the voltage pin manualy to A6.

Perhaps it is possible to implement this hardware like the other hardware boards like this in Config.h:

```
/********************       OSD HARDWARE settings      *********************/
//Choose ONLY ONE option:
//#define MINIMOSD                  // Uncomment this if using standard MINIMOSD hardware (default for 95% of boards) 
//#define WITESPYV1                 // Uncomment this if using Witespy V1.1 OSD, select this to correct for both swapped bat1/bat 2 and to also use alternative resistors / pinouts.  
//#define WITESPYMICRO              // Uncomment this if using Witespy Micro Minim OSD, select this to correct for swapped bat1/bat 2.  
//#define RUSHDUINO                 // Uncomment this if using Rushduino
#define KYLIN250PDB                 // Uncomment this if running on a Kylin 250 FPV PDB (Using A6 as VOLTAGEPIN)
```

And do something like this in Def.h:

```
#ifdef KYLIN250PDB
    #define VOLTAGEPIN    A6
#endif
```

ShikOfTheRa, 2016-03-07:
Yeah - absolutely - would need one to try it out and test though :) !!

what OSD software do they normally uses / come with ?

amoksworld, 2016-03-07:
I have one and testet it sucessfully last Night:

https://www.youtube.com/watch?v=f9G6EoGvSN0

The Kylin 250 original comes with openPilot and minoposd, but openpilot is not realy good for a racer (and I am missing some things like the Callsign on the OSD) so I managed it to flash cleanflight on the board. But then minoposd wont work (because of the protocoll thing) and I decided to switch to the best OSD software available - scarab-osd ;)

Now with cleanflight and scarab-osd this is a really awesome thing :D
I love this combination and the features they both give you now...

ShikOfTheRa, 2016-03-07:
The best OSD software available - scarab-osd - :~) lol....

If KDS models would send a kit they can be added to the list for sure  ;)
The ones that have been added have provided some gear to help with testing etc.

Looks nice bit of stuff although they probably should move from OP now - great software, but unfortunately not exactly popular anymore...

choyado, 2016-04-04:
amoksworld

how were you able to flash scarab-osd to the kylin?

amoksworld, 2016-04-05:
I have connected the Pins from the OSD on the PDB to a USB2TTL interface as described in the [OSD Setup Guide](http://www.kdsmodel.com/data/file/manual/kylin250osd_en.doc) and then - like on each other board - I used the [Arduino Software](https://www.arduino.cc/en/Main/Software) to flash scarab-osd to the chip ...

Before that I already flashed the FC on the PDB with CleanFlight, because openPilot was not supported by scarab-osd until now.

If you are interessted in the long Version of this story, have a look at [rc-heli.de](https://www.rc-heli.de/board/showthread.php?p=3103596#post3103596)

ShikOfTheRa, 2016-04-11:
Added

closed 2016-04-11 by ShikOfTheRa