ShikOfTheRa/scarab-osd #195: Heading based on Yaw-Axis (for Multirotor)

amoksworld, 2016-03-07:
Is it possible to implement a way to use the YAW-Axis information to show a Heading Rose and Heading Values instead of using GPS or MAG information?

I would prefer (or better wish) an option that locks the north in while arming, so you can align a copter without GPS or MAG manually before starting and use the heading informations.

It could be configurable with some DEFINE lines in config.h like it was already done for MAG:

`//#define USEMAGHEADING             // Undefine this to use MAG for FW heading instead of GPS (requires controller with MAG sensor) `

Perhaps something like

`//#define USEYAWHEADING             // Undefine this to use YAW for heading (for controller without MAG or GPS sensor) `

ShikOfTheRa, 2016-03-07:
Unfortunately not. :(

ShikOfTheRa, 2016-04-11:
closing this as cant implement a satisafactory solution. Sorry

closed 2016-04-11 by ShikOfTheRa