ShikOfTheRa/scarab-osd #197: MacOS X app not working.

karman-san, 2016-03-11:
I got and error saying 'can not run app' as of MWOSD_GUI 1.6.

Running on OSX El Capitan.

karman-san, 2016-03-11:
Found that permissions for some files inside MW_OSD_GUI.app are wrong, mostly missing execution bit. Tried to fix it by 'chmod -R +x MW_OSD_GUI.app'. Now I have a different error from OSX telling me that the app is damaged and should be trashed.

karman-san, 2016-03-11:
Found a way to configure the MWOSD by launching manually application.macosx/MW_OSD_GUI.app/Contents/MacOS/MW_OSD_GUI from the terminal.

ShikOfTheRa, 2016-03-11:
In the faq are some solutions for this

jwellman80, 2016-03-16:
I was having the same issue and rebuilt and exported with processing.  You can find the .app here:
https://www.dropbox.com/s/8vqslop50d9kudh/MW_OSD_GUI.zip?dl=1

ShikOfTheRa, 2016-03-17:
Can you let me know if it works for you

karman-san, 2016-04-06:
Sorry for the late response. Worked fine for me. Thank you both.

closed 2016-04-11 by ShikOfTheRa