ShikOfTheRa/scarab-osd #198: Can't write config to OSD from GUI. Simulate on OSD is ON.

c4pic, 2016-03-12:
I flashed my MinimOSD with R1.6. When i connect GUI to OSD , 'Simulte on OSD' is ON and then GUI is blocked and i can't change or write new settings.

I  changed the line    <entry key="Simtype">1</entry> from GUI.CFG file to <entry key="Simtype">0</entry> , then works fine.

ShikOfTheRa, 2016-03-14:
i intended to disbable but forgot. I need it on for development.
what FPS are you getting? Its usually a sign of a really old pc or something else hogging cpu power

c4pic, 2016-03-14:
I run GUI in a Win 8.1 64bits Laptop / Core 2 Duo T5550 1,83Ghz / 3Gb RAM

I'm getting about 4 FPS without 'Simulate on OSD'. With 'Simulate on OSD' enabled, GUI is blocked.

ShikOfTheRa, 2016-03-17:
I have repackaged the zip file to include it with that disabled by default.
4 fps should be OK.

Someone is building a chrome app right now to replace / provide alternative the the current GUI. No ETA, but hope it will be available for next release. He has made amazing progress with it

c4pic, 2016-03-17:
Thank you!! Great job!

closed 2016-03-21 by ShikOfTheRa