ShikOfTheRa/scarab-osd #201: Can't open GUI - OSX (10.11.1 15B42 / Java V8 Rev77 )

NicHume, 2016-03-29:
Very simply, the GUI will run, but won't display any windows... 

i.e. A dock icon appears, and I have access to a very limited task-bar menu, but the GUI window does not appear.

I've run the GUI on the same machine previously, but those previous version now don't seem to work either, so I can only assume that something that was updated in java or OSX has broken it? (And previous GUI builds?) 

Micro MinimOSD was successfully flashed to both MW_OSD versions 1.5 and 1.6 (GUI will not start for either.)

Running:
Macbook Air (13" / Early 2014) 
OSX 10.11.1 Build 15B42 
Java V8 Update 77

NicHume, 2016-03-29:
P.S. To be very clear: The app *DOES* appear to start/execute, and I do *NOT* receive any messages. It seems to simply not be displaying it's window.

ShikOfTheRa, 2016-03-29:
Maybe try the version compiled in here... 
https://github.com/ShikOfTheRa/scarab-osd/issues/197

It works OK on my hackmack running in virtualbox.
In future there will be a chrome version of the app.

NicHume, 2016-03-29:
Tried the version compiled for Issue #197  - Still the same result (The first time I tried running this build I was asked if I wanted to run software from an unknown developer. This was expected as it was a new download.) 

A Chrome version would be *AWESOME* ... I don't suppose there's an alpha or beta I could have access to, in order to try that is there? 

I may try firing up a virtual windows machine and see if I can run through that, but that's honestly a bit of a pain in the butt :/

NicHume, 2016-03-29:
If this helps you troubleshoot at all: 

I've got an old KV_Team_OSD_GUI build (v2.3) lying around, and it starts just fine. (I'm aware that they're different than MW_OSD, but I seem to recall MW_OSD is a fork of KV_Team? Or the other way around?) 

I've also got a bunch of MW_OSD versions dating back to 1.0 in a folder, and every one of them behaves the same and does not start / display it's window.

ShikOfTheRa, 2016-03-31:
Chrome unfortunately still in development. Not ready to use for real yet. 

Yes MWOSD is a fork of KVOSD. 

Unless want to run it in a virtuasl PC, you could download "processsing" for mac, load up the gui source (in "other" folder and compile your own - or just run it. Its very easy - interface similar to arduino

ohhorob, 2016-04-03:
I tried with Processing v3, but it complains about the call to "size()".

Tried with Processing v2.2.1 and it complained about missing controlP5 library.

Installed controlP5 from the menu "Sketch > Import Library... > Add Library". Searched for controlP5 and it installed.

Now, attempting to run gives the error "the function setControlFont(PFont) does not exist".

I'm on Mac OS 10.11.3, and java 1.6.0_65

$ java -version
java version "1.6.0_65"
Java(TM) SE Runtime Environment (build 1.6.0_65-b14-468-11M4833)
Java HotSpot(TM) 64-Bit Server VM (build 20.65-b04-468, mixed mode)

nathantsoi, 2016-04-03:
You need the exact dependencies listed here: https://github.com/ShikOfTheRa/scarab-osd/blob/master/OTHER/MW_OSD_GUI_SOURCE/MW_OSD_GUI/setup.sh

ohhorob, 2016-04-03:
Thanks. That works fine

ShikOfTheRa, 2016-04-21:
closing as resolution worked - thanks natahantsoi

closed 2016-04-21 by ShikOfTheRa