ShikOfTheRa/scarab-osd #202: Compatibility with Eachine 800TVL CCD camera

toto263, 2016-03-31:
Hi, I have a problem with MinimOSD and Eachine 800TVL camera. 
What happened is that I could see the video feed, but no overlay. If I unplugged the camera, I could see the overlay, but no video.
I checked all settings including NTST/PAL. Tried other cameras (600TVL, 1000TVL) and all of my cameras are working OK.
Only Eachine 800TVL CCD camera does not work properly.

ShikOfTheRa, 2016-04-01:
Its sounds liek a camera issue to take up with eachine.

Suggest can try DISABLING this...
#define USE_VSYNC                   // Disable if no screen display. Removes sparklies as updates screen during blanking time period.

seuffert, 2016-04-03:
having the same issue , same camera. tried every option including USE_VSYNC. nothing worked. no overlay :(

ShikOfTheRa, 2016-04-04:
yep - seems like its a common one which is incompatible with OSD's using max chip - I don't think there is anything we can do.

Suggest someone contacts |Eachine to see if anything can be adjusted on the camera?

seuffert, 2016-04-05:
I have contacted Eachine - with reference to this issue. Hopefully they will support us.

ShikOfTheRa, 2016-04-06:
OK so guess not much we can do...
Suspects its their issue by producing a camera that provides out of spec signal..... Just a guess though.. 

they aren't teh only one unfortunately. If you check runcam's site they actuall do specify in red when a camera is incompatible with OSD.

BoyDrone, 2016-04-16:
I'm experiencing the same issue on my end. Its kind of frustrating to deal with when there is no specification to match up. also kind of a waste of money when you think about it when people buy cameras that dont work. Is there a mini minimOSD camera compatibility list anywhere? that in itself would be fantastic and save a lot of trouble. How do we know that the next camera we purchase will or will not be compatible with this board?

ShikOfTheRa, 2016-04-16:
Not aware of any such list. I can advise the runcam swift I just purchased is compatible, small, light and so far appears to be quite a decent cam. Should have a flight vid in next couple of weeks.

toto263, 2016-04-17:
I confirm the following cameras work with MinimOSD.

* Eachine 700tvl CMOS FPV Camera 
http://www.banggood.com/Eachine-700tvl-13-Cmos-148-Degree-Lens-FPV-Camera-p-972617.html

* Eachine 700TVL CMOS Mini Camera 
http://www.banggood.com/Eachine-700TVL-1-3-CMOS-2_6mm-Lens-148-Degree-Mini-Camera-p-991028.html

* 1000TVL FPV HD CMOS Camera
http://www.banggood.com/1000TVL-FPV-HD-COMS-Camera-2_8mm-Lens-120-Degree-for-Multicopters-p-974382.html

seuffert, 2016-04-17:
Also working:

- Andoer  700TVL Mini FPV Camera with 1/3 "SONY Super HAD CCD II CCTV FPV Camera 2.1mm Lense (Sony Effio-e 700TVL Kamera)
http://www.amazon.de/Andoer-700TVL-Kamera-Super-Effio/dp/B00QJRUHWO

toto263, 2016-04-17:
I did not test the 110 degree Eachine cameras, but I think these work also because I had no issue with 148 degree cameras.

* Eachine 700TVL CMOS Camera 110 degree
http://www.banggood.com/Eachine-700TVL-2_6mm-Lens-1-3-CMOS-110-Degree-FPV-Mini-Camera-p-991027.html

* Eachine 700TVL CMOS Mini Camera 110 degree
http://www.banggood.com/Eachine-700TVL-13-Cmos-110-Degree-Lens-FPV-Camera-p-972616.html

BoyDrone, 2016-04-18:
I found this: http://www.rc-cam.com/forum/index.php?/topic/4003-minimosd-software-fix-improve-camera-compatibility/&page=2&_fromLogin=1

but I cant seem to figure out how to implement this ... keep getting errors.

ShikOfTheRa, 2016-04-18:
That looks to appear to solve a different issue. I think its already included in MWOSD

ShikOfTheRa, 2016-04-18:
From what I can tell so far cameras that have Nextchip 2090 DSP seem to be the main ones affected.

ShikOfTheRa, 2016-04-18:
The popular Nextchip 2040 is OK....

closed 2016-04-18 by ShikOfTheRa