ShikOfTheRa/scarab-osd #203: Feature request: implement voltage percentage

nayebzadeh, 2016-04-03:
Hi.

Thank you very much for the cell count detection feature! This is a killer feature for me! One little thing missing to make this feature perfect though.

It would be great if you could besides of the actual voltage value also display the voltage percentage based on the voltage alarm setting. 0% meaning the voltage alarm value is reached and 100% being 12.6V for 3S and 16,8V for 4S.

Maybe someone can add this feature in the future.

Thanks in advance.
Human

tracernz, 2016-04-03:
The relationship between capacity and voltage is non-linear so a linear percentage like this will be misleading.

nayebzadeh, 2016-04-03:
This is correct. 

But as a pilot you always have to stop flying when a certain voltage is reached. That voltage (alarm level) could be the reference for 0%. 

It should not represent the percentage of the battery capacity but the percentage of flight time you still have. 

A lot of lipo testers show a more or less accurate percentage of the remaining capacity until you should stop flying with that battery. 

It would be like that. You could of course call it "Remaining flight time percentage" which is based on the alarm level of the voltage and maximum voltage. 

Cheers,
Human

ShikOfTheRa, 2016-04-04:
Thanks for the feedback

There is the battery evo icon - not quite a % value.

I'll add it to the wish llist. 
Also is probably a horizontal or vertical bar similar to the Vortex OSD

Unfortunately its not on the todo list yet.

ShikOfTheRa, 2016-04-21:
I am going to close this as its been added to teh development wishlist/roadmap (in other / documentation / roadmap

closed 2016-04-21 by ShikOfTheRa