ShikOfTheRa/scarab-osd #204: Cell count detection question

skaman82, 2016-04-03:
On the release notes I've read that mwosd supports autodetection of cells. I usually fly with 3 AND 4S lipos. I've configured MWOSD this way
[image]

If I connect a 3S battery, I get a low voltage warning. Should it not detect now a 3 cell lipo now? And the voltage alarm setting should be a voltage per cell. Did I something wrong with the setup? How does this work?

ShikOfTheRa, 2016-04-04:
FAQ #26 in FAQ. Just updated for you
https://github.com/ShikOfTheRa/scarab-osd/blob/master/OTHER/DOCUMENTATION/FAQ.md

skaman82, 2016-04-04:
@ShikOfTheRa Thanks! I'll give it a try

skaman82, 2016-04-16:
works like a charm, thank you. It would be better to activate this feature in the GUI tho :)

closed 2016-04-16 by skaman82