ShikOfTheRa/scarab-osd #205: Osd locks up

islander46, 2016-04-06:
I am using a naze 32 full with base flight and an os doge. Everything works fine until I arm the flight controller. Then the osd says no data disarmed and the osd freezes. When I disarm the osd unfreezes. I have tried using a bec to power the VTx and camera, just in case it was a power problem. No luck, same problem. The only thing I have done differently from other builds that work fine is, I have enabled the gimbal function and bumped the ESC signal wires to outputs 3-6 on the naze. It am beginning to think I have a bad os doge... Any Ideas? 
P.S.
 I have reflashed the  osd, no help.
 I am not using a gimbal anymore but the ESC signal wires are still In 3-6.
Newest stable versions of all firmware

u8mydog, 2016-04-07:
That sounds like what happens when the same serial port is used for configuration the controller board and for OSD.The flight controller will not send info to the OSD until the flight controller is armed. This way, if the controller is not armed, you can connect to it via usb (or whatever) and access the controller's configuration.

islander46, 2016-04-08:
That was it. I disabled fr Sky telemetry and it started working.
I miss having the voltage spoken to me by the radio is there a way to still have telemetry?

u8mydog, 2016-04-08:
Get an F3 controller that has more uarts :) 
Or use a softUart for OSD only.Or use a softUart for configuration but then you can't use the USB on the FC. You would have to use bluetooth or a FTDI (usb to serial) module plugged into the softuart pins for configuration.

ShikOfTheRa, 2016-04-11:
Thanks for support....

closed 2016-04-11 by ShikOfTheRa