ShikOfTheRa/scarab-osd #207: Eachine CCD 700TVL camera flickering

prapdm, 2016-04-08:
Hi,
I have problem with this camera.
http://www.banggood.com/Eachine-CCD-700TVL-148-Degree-Lens-FPV-Camera-p-977401.html

Text from OSD is flickering. How to resolve this?

ShikOfTheRa, 2016-04-11:
get a different camera....
Unfortunately its one of the ones that seems to be incompatible with MAxim chip based OSD's

If Eachine send me some samples I coupld have a look. It has been reported to them. Its one of teh other open rissues

prapdm, 2016-04-11:
Not good. Flickering is a little bit anoing. I tried add some resistors to between osd and camera without success. I also tried USE_VSYNC and FASTPIXEL - no success. Is it possible to do some OSD software changes (adding more compatibility) or it is hardware issue ?

ShikOfTheRa, 2016-04-11:
I think its the camera sending out of spec syncs. 
I dont think there is anything that can be done.

Best success I have had is to use lareg or bold font. Its not ideal...

ShikOfTheRa, 2016-04-21:
No appropriate response from Eachine. Nothing can be done here as its down to cam/chip manufacturers so going to close unresolved. Sorry couldn't do more...

closed 2016-04-21 by ShikOfTheRa