ShikOfTheRa/scarab-osd #212: Flickering of OSD on "Westerlund 32"

DasSmeagle, 2016-04-21:
Hi,

I just flashed MWOSD R1.6 on my "Westerlund 32" and I can hardly see/read the OSD now. Its flickering... The Westerlund is a Naze32 + MinimOSD on one board. 

It originally came with KV_OSD 2.2 and OSD was working fine. So I'm quite sure it's not incompatible camera like mentioned in the FAQ... ;)

This ist the controller: 

http://www.goodluckbuy.com/star-power-westerlund-32-flight-controller-with-osd-gyroscope-for-rc-multicopter.html

It was included with a "Jumper 218 pro" race quad... 

http://www.hobbyking.com/hobbyking/store/__82905__Jumper_218_Pro_Quad_Copter_ARF_.html

Any idea? My big Problem is, I can't flash back to the original firmware, because I always get several error when I try to flash the "KV_Team_OSD.ino"... And I can't start the KV_Team GUI...

I can flash MW OSD without problems, GUI for configuration works, too. But the OSD is hardly readable because of flickering....  :( If I concentrate hard I can read the OSD, and apart from flickering it seems to work just fine...

Help!

Thanks,
- Oliver

DasSmeagle, 2016-04-21:
Update:

I just tried to toggle this setting, but it didn't really help:

#define USE_VSYNC                   // Disable if no screen display. Removes sparklies as updates screen during blanking time period.

Toggling this changed the "style" of flickering a little, but still not good.

- Oliver

ShikOfTheRa, 2016-04-22:
can you show a vid so I can see type of flicker?

DasSmeagle, 2016-04-22:
Wow, this is strange.

I just tried to take that video for you. Today I had NO OSD at all. So I wanted to try what happens when I disconnect the camera, which is not so easy on this micro copter, had to disassemble part of the chassis.

After disconnecting camera, the OSD was working, as espected. OSD on black screen.

Now the strange part: After connecting camera again now the OSD works, too! No flickering, perfect.

Even after rebooting, everything is fine!?!?! I have no idea, what happend. The connector is only for power, ground, video. So any of those 3 connections missing, nothing should work. And bad connection for video is unlikely too - the picture from the camera was fine, just the OSD was missing.

Sorry for bothering you, I think I am fine now.

Thanks!
- Oliver

ShikOfTheRa, 2016-04-22:
Result! Lets hope it stays
Note it will flicker if the 5v to to OSD is low. I have seen that only once

closed 2016-05-03 by ShikOfTheRa