ShikOfTheRa/scarab-osd #220: Master is broken with #213 (MAVLINK telemetry support)

jflyper, 2016-04-26:
The master no longer works after #213 (MAVLINK telemetry support).

Environment:
OSD: MinimOSD/MicroMinimOSD compat setup (Arduino Nano + Sparkfun MAX7456 breakout)
FC: Naze32 compat (FLIP32AIO)

Config:
Stock master.

Symptom:
Opening screen good, no data/disarmed display good.
Talks with GUI fine.
As soon as it start communicating with FC, it momentarily (for a few, if not one frames, as it looks) displays what it should display(sidebars and such), and goes to blank.
Disconnecting and reconnecting FC momentarily reveals voltage display at the top left corner, suggesting sync parameters may be okay.
The voltage is blinking, and is quickly cycling when visible.
The voltage display goes a way quickly, the display stays blank until next disconnect/reconnect.

jflyper, 2016-04-26:
Okay, I think I've found the problem. See #221.

@ShikOfTheRa 
Can you fix this?

ShikOfTheRa, 2016-04-26:
yep done thanks !

closed 2016-04-26 by ShikOfTheRa