ShikOfTheRa/scarab-osd #225: Arduino IDE 1.0.6?

jflyper, 2016-04-29:
I just came to notice this in the OTHER/DOCUMENTATION/developer guide:

> Arduino version:
MWOSD must be able to compile with Arduino 1.0.6.
Newer releases ususually provide smaller optimised code. Consider a benefit.<br>

Conversely, what is the benefit of staying within the limit of 1.0.6?

I'm foreseeing MWOSD going multi-platform in the near future, and newer releases generally provides better code portability. Why not consider migrating?

nathantsoi, 2016-04-29:
that must be out of date, it compiles fine w/ the latest arduino studio for
me

jflyper, 2016-04-29:
@nathantsoi 
I was thinking the other way around;  backward compatibility of some sort, including host environments.

nathantsoi, 2016-04-29:
hum... i guess i dont get it. we want to continue to support 1.0.6? why
would we want that?

jflyper, 2016-04-29:
That's my original question :)

nathantsoi, 2016-04-30:
gotcha. yeah, im down to drop arduino 1.0.6, but i also have no idea why we'd need it in the first place?

i've been using this makefile w/ arduino builder: https://github.com/nathantsoi/scarab-osd/blob/new-platform-prep/Makefile

tracernz, 2016-04-30:
Because supporting more versions == less support headache?

ShikOfTheRa, 2016-05-03:
Why Arduino 1.06?

Well at some point I upgraded to a newer one and released some code that wouldn’t compile. I wasn’t prepared for the volume of issues it created. Filled up my pm box….

It would appear that many people still uses 1.06 and earlier because:
Some just haven’t upgraded
A bigger issue is some are forced to stay with it because other code and libraries hasn’t been updated to work with 1.6 (had to do same for MWOSD to run on 1.6

1.6 is significantly more optimised and compiles to a smaller footprint. 
So having it compile and run on 1.06 means no issues with other versions. And I get less PM’s

ShikOfTheRa, 2016-05-03:
Development environment for MWOSD.

Firstly – it would be my personal preference to move away from Arduino to a better environment like eclipse etc.
This may seem to hinder any advance, however the environment is to remain Arduino. 

This is my reasoning:
- Overriding factor – most users / contributors are not professionals and Arduino brings a certain ease of installation and ability to amend.
- MultiWii moved – and code contributions all but stopped. It is now a dead project.
- KVOSD similarly moved and is in a similar position.
- I like that people can use this project also as an easy way of learning and start hacking / tweaking and ultimately contributing.

I don’t want MWOSD to go down the same path as the others above.

If this were STM with enough memory where we could compile a single hex file and use a GUI /CLI like baseflight / cleanflight etc. then that would create a different answer. 

Apologies to the frustrated developers for this statement. I am impacted too… however I wish this project to remain open to all for modification. AND I don’t want a gazillion emails asking how to set up eclipse or some other environment...!!

jflyper, 2016-05-03:
@ShikOfTheRa 
I really appreciate the in depth explanation.

Sticking with the Arduino IDE is no problem. I do have some thoughts on the release,  but there's no strong points at this moment.

ShikOfTheRa, 2016-05-03:
I wish it was a better answer. I really do...
In hindsight it would have been so much easier to start from scratch rater than to keep hacking / amending a base which was never really designed t deliver hat it does today.

closed 2016-05-04 by jflyper