ShikOfTheRa/scarab-osd #228: OSD sending MSP request 0x00

jflyper, 2016-05-03:
Current master (dc87594) periodically transmits erronous request 0x00. Interval between the requests is around 150msec, and is sometimes back to back with another valid message, so it looks as if it is timer driven.

Is this intensional, say, for heart beat or something?

Request at 47+msec, response at 49+msec.
[image]

Request zoomed up
[image]

Response zoomed up (error response)
[image]

Back to back with another valid request
[image]

ShikOfTheRa, 2016-05-03:
Not a rerquirement I am aware of. 

I'd be inclined to look at modeMSPRequests and queuedMSPRequests. Something funcky there I suspect

Im guessing the command scheduler is empty it sends nothing until it is reset. Can't see why at a quick look

Be inetersting to try this and if that it, work backwards
Change to
    if(!fontMode){
      #ifndef GPSOSD
      if (MSPcmdsend>0)
        mspWriteRequest(MSPcmdsend, 0);

ShikOfTheRa, 2016-05-03:
What tool is that BTW - looks like the sort of thing could do with!!

jflyper, 2016-05-03:
Yeap. I had just tracked down the problem to the section of the code you are pointing at.

By examining and bypassing the MSPcmdsend==0 case, the offending request will go away, but I'm not sure if this is the right correction; need to understand what queuedMSPRequests and modeMSPRequests are doing.

jflyper, 2016-05-03:
The tool is the Saleae's logic analyzer. Mine is the Logic 8. For $219 + S&H, I believe it's an incredible value. I just can't imagine my life without it!!!

https://www.saleae.com/

ShikOfTheRa, 2016-05-03:
OK, so in the correct area.

I wonder if there is an MSP req not covered by the SWITCH statement or the queue is empty Both will send a zero. And neithe rshould happen :)

Suggest add in an ELSE... and debug 
#ifdef HAS_ALARMS
      case REQ_MSP_ALARMS:
          MSPcmdsend = MSP_ALARMS;
      break;
      case ELSE
          MSPcmdsend = MSP_ATTITUDE;
         debug[0]=req;
      break;
#endif
    }

ShikOfTheRa, 2016-05-03:
Tool looks awesome. Beyond my financial means :(

jflyper, 2016-05-03:
Okay, the problematic case is: req = 64 (REQ_MSP_ATTITUDE), which is only processed  in the MSP_SPEED_LOW configuration.
```
 323       case REQ_MSP_COMP_GPS:
 324         MSPcmdsend = MSP_COMP_GPS;
 325         break;
 326     #ifdef MSP_SPEED_LOW
 327       case REQ_MSP_ATTITUDE:
 328         MSPcmdsend = MSP_ATTITUDE;
 329         break;
 330     #endif //MSP_SPEED_LOW
 331       case REQ_MSP_ALTITUDE:
 332         MSPcmdsend = MSP_ALTITUDE;
 333         break;
```

We can #ifdef the inclusion of REQ_MSP_ATTITUDE into the modeMSPRequests:
```
 709      #ifdef HAS_ALARMS
 710       REQ_MSP_ALARMS|
 711      #endif
 712 #ifdef MSP_SPEED_LOW
 713       REQ_MSP_ATTITUDE;
 714 #endif
 715     if(MwSensorPresent&BAROMETER){ 
 716       modeMSPRequests |= REQ_MSP_ALTITUDE;
 717     }
```

There is another one for configMode case:
```
 667   else if(configMode) {
 668     modeMSPRequests =
 669       REQ_MSP_IDENT|
 670       REQ_MSP_STATUS|
 671       REQ_MSP_RAW_GPS|
 672 #ifdef MSP_SPEED_LOW
 673       REQ_MSP_ATTITUDE|
 674 #endif
 675       REQ_MSP_RAW_IMU|
 676       REQ_MSP_ALTITUDE|
```

I don't know the effect of the latter; gonna test it.

jflyper, 2016-05-03:
Desktop test done. See #230 

P.S. I've realized that the attitude is not even used in config screens. May be drop it from the configMode case ...

ShikOfTheRa, 2016-05-03:
Ah yes, I see the issue. That should do the trick. And agree config mode not required...

closed 2016-05-04 by jflyper