ShikOfTheRa/scarab-osd #232: Strange issue with micro minimOSD – already three OSDs dead

skaman82, 2016-05-08:
I know, it is kind of a support question, but I am googleing all day and find nothing and hope you've encountered this before. I have build two new quads to fly with 4s and 3s batteries. I've build in am micro minimOSD with MWOSD and activated the autocell feauture. The OSD is hooked up on a separate 3A BEC and a 5V cam is powered from the OSDs 5v pad - like an all my other quads. 

The OSD work ok for some days, but when I arrive at my flying field I see just a black image (no video and no OSD overlays). After removing the OSD i noticed that the green light ist blinking differently then before. Now there is a pause in between the blinking sequence. I've tried to clear the eeprom an flash MWOSD again - that works just fine. The OSD connects to the GUI normally. But the blinking pattern with the pause still remains and I see the same black image. 

This happens to me third time already in a very short time and I am getting tired to buy new OSDs and replace them. So I have the following questions:

1. Do you encountered this before and how I can fix it? Never had the problem on my other quads with osds. The only difference ist autocell feature and 4s/3s batteries. Already checked different BECs with a multimeter. They all spike to about 7.6v for a second when I plug in the battery and then deliver constant 5v after that.

2. Can I save the strange blinking OSD somehow or should I just replace that?

Thank you

skaman82, 2016-05-08:
Added a video of the strange blinking pattern: https://www.youtube.com/watch?v=z2ZQbgHEvbw

jflyper, 2016-05-09:
@skaman82 
I am very sorry to tell you that I think your OSDs, more specifically MAX7456 chips are all dead, probably due to rush voltage/current. If the current/voltage last long enough to be observed by a multimeter for a second, then the actual duration is much longer, and the actual peak voltage is much higher.

The green LED on Micro MinimOSD is connected to SCLK signal of SPI bus, and keeps blinking very fast as MWOSD is constantly redrawing the entire screen. Irregular blinking, as seen in your video is observed when MWOSD detects a stalled MAX7456 and trying to reset it. In your case, it's constantly trying to reset a dead MAX.

Why do ATmega328s survive and MAX7456s do not? Theory is that ATmegas are more tolerant against spikes than MAXes and Clones. See a FLIP32AIO thread in the regroups for lots of dead MAXes.
[http://www.rcgroups.com/forums/showthread.php?t=2535914](http://www.rcgroups.com/forums/showthread.php?t=2535914)
In the thread, people came up with protection against spikes using capacitors and/or zener diodes.

@nathantsoi 
Any comments?

skaman82, 2016-05-11:
@jflyper thanks. I tested some popular BECs and PDBs - they all have a small voltage spike when I plug in the battery (3s or 4s) - so it is not that I think, otherwise almost everybody would have this problem. Could it maybe have something to do with spikes produced by littlebees ESCs with damped light in flight? How should I hook up the cap or diode to my minimOSD? Will this diode do? Fairchild BZX79C5V6 or is a cap better?

jflyper, 2016-05-15:
@skaman82 The spec look okay to me. Here's some one using the identical spec zener:
[http://www.rcgroups.com/forums/showpost.php?p=33807477&postcount=827](url)

skaman82, 2016-05-15:
@jflyper thanks for the link. In that example it is hooked up to a different board. I am not so good in eletcronisc so i am not sure how to hook it up to my Micro minimOSD board. just between + and - lead? Won't this create a short?

jflyper, 2016-05-15:
Solder it to any Vcc/GND, with 'Band'ed side on Vcc.

With this hookup, if the diode was a regular diode, it just do nothing; it blocks current to flow from Vcc to GND. The magic of zener diode is that this blocking function only works when voltage across the diode is lower than its break down (zener) voltage, which is 5.6V for your choice. ANY and ONLY 'excess' voltage will be drained (by letting the current flow through the diode). You can think of the effect as bypassing, rather than shorting the circuit.

I looked for some cartoon explaining the above on the net, but couldn't find one, so made up my own. It is not technically correct, but gives you the idea of the 'quantified bypassing'.
[image]

chickenman2222, 2016-05-19:
I am soooo tired of this problem. I have lost video and OSD data on 7 different micro minims.  Most happen at battery plug in after the third 20 minute flight, but my last one happened about 7 minutes into the third flight, during flight.  I cant link to some of the info above.  What diode should I use. I have a rectifier diode 1n4001 will that work? if not which diode should I use. Where can I buy a better minim?  Obviously I like the minim and want to stick with it. What are the ATmega 328's? Is that the minim?  The following is my current setup 
[image]

Arakon, 2016-05-21:
The diode has to be a Zener, otherwise it won't do anything.

chickenman2222, 2016-05-21:
Yeah, thanks When I was first researching this issue some one had stated 1N4001 number. I didn't realize at the time that it wasn't a zener diode . Since then I have ordered the BZX79C5V6 and am awaiting delivery.  Thanks for your input.

Rodrigoah, 2016-07-12:
I've also had two micro minimosd die on me... ¿Any chance of revival?

jflyper, 2016-07-12:
How is your green LED blinking?

It should be continuously blinking very rapidly.

Watch carefully; with a dead MAX chip, the rapid blinking hicks-up (momentary stops) once every second.

chickenman2222, 2016-07-13:
I have a new updated diagram,  The diode didn't help. The capacitor didn't help.  You can't use a 5 v BEC or it will kill the minim. You can get the 5v needed for the minim power from the FC card or you can order a power hub and get it off of that. Plus it is very important to only use one ground from minim to quad. I don't use the 5v ground, instead I use the 12v voltage monitor ground only..... I do also use a video ground from the camera to the minim, not from the minim to the transmitter.
[image]

jflyper, 2016-07-13:
@chickenman2222 You don't have to post here if you post to fpvlab; I'm monitoring both.
@skaman82 Is zener working?

Rodrigoah, 2016-07-13:
@jflyper 

Here are both of my micro minim osd, sorry about the frame rate of the second half 

https://youtu.be/mjNFpA7xO-Y

jflyper, 2016-07-13:
@Rodrigoah Are these both fried? The first one looks okay to me. The second one, is this another micro minim? It's really hard to tell :(

skaman82, 2016-07-13:
@jflyper I managed to get this under control with a 1000uF 25V Low ESR cap. I just soldered the cap to my PDB and had no burned OSDs since then.

jflyper, 2016-07-13:
@skaman82 1000uF!!! Cool!!!

skaman82, 2016-07-13:
@jflyper https://www.youtube.com/watch?v=ZhvuQJPGHzU

Rodrigoah, 2016-07-13:
sorry @jflyper here is another video of the second minimosd 

https://youtu.be/_PBMQVRR3-w

I've also tried checking the input voltage spike of the 5v BEC and it seems there isn't one

[image]

Both my OSD are dead...

jflyper, 2016-07-13:
@Rodrigoah Ah, the second one is dead as a hyper cold lunar stone.
And the first one, I reviewed the video clip again, and noticed the "breathing", in the 3~4sec section. If you could see it in the video, then you can find it live: breathing once every second.
I suppose the scope trace is from initial power application, right? See the video clip @skaman82 pasted above: [https://www.youtube.com/watch?v=ZhvuQJPGHzU](url)

@skaman82 It's a great video!!!

Rodrigoah, 2016-07-13:
@jflyper I've seen the video, but i'm not having trouble with noise (I used to have the OSD flickering on throttle up, it took me a while to solve noise problems). 

Yes, the scope is from initial aplication of power, then, the ripple is bearly visible even under throttle conditionsñ

¿Should the MAX chip throughput the input video when not having power?

u8mydog, 2016-07-13:
FYI: I blew a mimumOsd when I left tx/rx connected to the flight controller and then plugged usb into flight controller. I'm guessing that the OSD didn't like seeing active signals on RX when it wasn't powered up. I power the OSD from a separate BEC and not from the FC.

chickenman2222, 2016-07-13:
@jflyper No the zener diode didn't help. I even tried the zener and the capacitor together, and separate and no help, Finally I took the 5v power off the FC board and it works. I have another I used a power hub for the 5v and it works too. The minim does not like a BEC.

jflyper, 2016-07-14:
@Rodrigoah The point of the video @skaman82 pasted is not only the noise, but the voltage spikes and dips that ranges that sometimes shoots up to +29V. Although the video was intended to provide a cure for Lilbee's back rush on active breaking, this kind of phenomenun is common, so we have to prepare for it. For FC's tx/rx while not powered up, these are connected to ATmega328P, so should be okay but who knows.

ShikOfTheRa, 2016-07-21:
Think we are done here. 

I'll add to the FAQ a section for power. Propose something like the following :
Poor quality voltage supply can significantly impact the the reliability of your the system. This ranges from high voltage spikes that will cause your FC or OSD to fail completely to low voltage brownouts that will cause your FC or OSD to hang. All are bad, but all can be avoided.
Regenerative / Braked ESC such as those using BLHeli / Damped light option contribute significantly to the issue.
Systems with digital servos powered through the FC are also vulnerable.

It is important to provide a clean supply to your FC and OSD etc. Consider 5v and 12v connections (and also for FC/OSD with embedded switching regulator , the battery supply). Also consider any servos connected to the 5v on the FC.

Options to consider in order of recommendation of least effective to most effective are:
  0 Use a good ground wiring layout design to a common point (consider this mandatory)
  1 Use a high value electrolytic cap on supply lines (3300uf, 16v for 4s)
  2 Use a Low ESR, high value cap on the supply lines (1000uf, 16v low ESR for 4s). 
  3 Use a traditional inductor / capacitor (LC) filter on the supply lines to the electronics
  4 Use a linear voltage regulator circuit (7805/7812 etc)

For overvoltage protection:
  0 Consider use of a protection diode (BZX79C5V6)
Gary Moscardini

jflyper, 2016-07-21:
Nice write up. I consider it as a BCP. +1

ShikOfTheRa, 2016-07-21:
OK - added to FAQ !

flounderscore, 2016-07-21:
16V is not enough for 4S. Especially with potential higher back-EMF I'd use
at least 25V caps for larger safety margin. I have seen many 16V (tantalum)
caps blow up on a 15V circuit.

ShikOfTheRa, 2016-07-22:
I've experienced that too with tantalum caps. usually blows the head off completely. I always use ones with 50% higher voltage rating. Probably a bit old school 

Technically I'd agree on 25v for 4s, but for the relatively low run time, electrolytic's ability to handle breakdown better and the larger can size personally I'm OK with that. I'll amend with 25v= better or something like that.

Really this is an issue for the hardware manufacturers though - not MWOSD

ShikOfTheRa, 2016-07-26:
I am going to close this as its an issue for hardware vendors rather than any other the MWOSD devs.

Its still searchable by google and documentation updated to reflect findings so far. Thanks to all for contributing

closed 2016-07-26 by ShikOfTheRa

nathantsoi, 2016-07-30:
sry, I know this is closed, but for anybody thinking about getting new microminim osd hardware, i just saw that one of my favorite manufacturers, airbot, has posted a version of the micro minim osd with proper power management and the upgraded AB7456 (I think they make this chip themselves) OSD chip: http://shop.myairbot.com/index.php/microosdv23.html

i think they also distribute through a bunch of other sites, so if you see the AB7456 chip on anything, it's probably a really good board to use b/c of the low power consumption and high tolerance to voltage spikes, basically all the lessons learned from this: https://nathan.vertile.com/blog/2015/11/09/rctimer-oze32-integrated-flight-controller-review-and-setup/, but in minimosd :)

anyway, hope that's helpful. keep up the great work all!

tracernz, 2016-07-30:
AB7456 is probably a Chinese clone, MAX7456 is no longer available from Maxim (it is 2016 after all). The problem is not with the MAX, but feeding dirty voltage straight to it, and is overcome by many existing designs, just not the widely available MinimOSD.

Saxin, 2017-03-14:
Sorry to revive this from the dead...

Which design are more robust to voltage spikes ? 
(I've killed two minium osd in two days, powered thru the FC serial port connection.... or maybe this is the issue, I have UBECS on my PDB one supply 5v and another at 9v. )

macfly1202, 2017-03-14:
@Saxin : I've burn 1 micro OSD with battery monitoring port despite voltage divider integrated to the board. I was far from max voltage of voltage divider of micro osd (was runnig on 3s) On other side, I initially started with original 3DR minimosd wich is far better on image quality/stabily with overheating. Side to side comparison with micro OSD show artefact in image if temperature is to high - genuine minimosd hasn't got these artefact. Micro minim osd is heating a lot for so tiny puppet. I've made hundred of hour of ardupilot with minimosd genuine without any issue. On some powerful racer I m using micro osd with polulu 5V step-up step-down S10V4F5. Awesome result against spike issue - reliabily. Don't know if these comments helps. I think like Nathan Vertile said more comments up : question of power design (Airbot is a reliable supplier).

Saxin, 2017-03-14:
@macfly1202  ... and you feed it Bat. voltage ?

ShikOfTheRa, 2017-03-14:
I have used a lot od OSD's but must admit I have not had any burn out apart from my fault when connecting.

I have had these made which are very resilient due to design....
They can be use din place of normal minimosd
https://www.virtualpilot.co.uk/index.php?route=product/product&product_id=51

Saxin, 2017-03-14:
Does the airbot osd compatible with MWOSD ?

macfly1202, 2017-03-14:
@saxin : only issue with micro OSD and abttery monitoring. you could make an external voltge divider with high quality resistor and unsolder + shunt mini resistor. If you are no limited to space, I suggest to get the normal original design size like one of ShitoftheRa suggested. There are all compatible to MWOSD has far as I know.

Saxin, 2017-03-14:
I'm not using any sensor on MinimOSD all is fed from FC, I have bat connected to FC + current sensor on PDD.

macfly1202, 2017-03-14:
which 5V are you using on micro ? the one between Rx Tx and Gnd Or the main entrance 5V - the bigger pad ? (I'm always using the big one - the tiny next to Rx only for Flashing firmware.  
[image]

Saxin, 2017-03-14:
I'm wired exactly as shown + gnd from cam and VTX to OSD.....
(I'm like so also on my two big copters on Ardupilot without any issues, but that is another story I guess)

ShikOfTheRa, 2017-03-14:
Best wiring model is as follows.

Avoids ground loops and differing potentials
Running grounds and supplies between/through components is not optimal
Not always required...
Also add LC filters in appropriate places
Its mode hassle. I don't always do it...
[image]

Saxin, 2017-03-14:
The scheme above you mean ?

Saxin, 2017-03-14:
I see now , thanks.
Still not clear regarding the VIN , and pads on the micro .... I was sure they are same pads maybe they aren't and there is some filtering after the main pads ?

ShikOfTheRa, 2017-03-14:
Vbat1 and vbat2 are the only pads that should have greater than 5v applied.

Vin = video in
Vout = video out

Saxin, 2017-03-14:
No no ....

Sorry I ask regarding the voltage in , big pads on the right or pads near Rx/tx ?

I don't have any other sensors connected to micro , only video side and Tx / rx