ShikOfTheRa/scarab-osd #234: Dronin support

dronophobia, 2016-05-10:
In the osx version of 1.6 i don't see dronin in config.h 
I used Taulabs msp instead, this gives inconsistent altitude.

Is support for Dronin coming in a next release?

Thank you very much for your valuable time.

ShikOfTheRa, 2016-05-11:
dronin and tau are identical as far as mwosd is concerned.

Does your FC have baro, gps or both. And is this fixed wing or multirotor?

dustin, 2016-05-21:
What is inconsistent in altitude?  They should be the same, though the values may be incorrect (or your altitude may not be valid).

Regardless, TL vs. dRonin doesn't make a difference here.

ShikOfTheRa, 2016-05-26:
Will close this as duplicate of  #245 which contagins more info

closed 2016-05-26 by ShikOfTheRa