ShikOfTheRa/scarab-osd #235: MSP_PIDNAMES not implemented

mindnever, 2016-05-12:
I have pending PR with LibrePilot (same family as Tau and dRonin) that brings dRonin MSP implementation there, with some improvements, specifically PID tuning. However, LP (TL and DR also) have slightly different set of tuneable PIDs and would greatly benefit of having MSP_PIDNAMES implemented so it can present own set of names to display.

For example, these are pidnames that are (to be) sent from LP:

```
static const char msp_pidnames[] = "ROLL;"
                                   "PITCH;"
                                   "YAW;"
                                   "A.ROLL;"
                                   "A.PITCH;"
                                   "PosR;"
                                   "NavR;"
                                   "A.YAW;"
                                   "MAG;"
                                   "VEL;";
```
(where MWOSD deliberately skips over PosR and NavR, and MAG and VEL are currently unused in LibrePilot)

https://bitbucket.org/librepilot/librepilot/pull-requests/232/lp-291-port_uavomspbridge_from_dronin/diff

ShikOfTheRa, 2016-05-12:
Hi - would be great to have librepilot support. let me know when it sin there and I 'll add in a Flight Controller choice - at least for the items supported today. 

Note MWOSD skips over simply due to lack of menu space.  

I have wondered if one possible way to adjust all items would be to maybe just display one line and use the tx sticks to change the line displayed.

Happy to consider any pull requests. As ever we have very little memory - so need tome #defines to disable when using other FC choices.

mindnever, 2016-05-12:
Pidnames support would take around 60 bytes of RAM, give or take few, considering that there is space for 7 chars per name only, so no point in allocating more than that.

ShikOfTheRa, 2016-05-12:
I am fully tied up with request at the moment. If want it in near future you are welcome to submit a PR. Otherwise I'll stick it on the wishlist.
It would be great to have it in here. Really great.

flash is full in some scenarios - less than 60 bytes left!. Its long overdue a tidy up though.
Stack is great now jplyer submitted some PR's to use flash.

mindnever, 2016-05-14:
I have submitted pr. It turns out that resulting code and additional memory usage is minimal.

closed 2016-05-16 by mindnever