ShikOfTheRa/scarab-osd #241: Overlay initialization problem.

dustin, 2016-05-21:
I've had this issue on both OSDoge using analog RSSI and microminim trying to use vbat 1.  I'm not entirely sure that's related as various other arrangements have been failing inconsistently for me tonight.

I've been toying around with it and found a way that consistently cause it to start up -- unplug the camera:

https://gfycat.com/CookedOldfashionedGoat

This seems to immediately fix the issue and shows a timer that's been running, adc values correct, etc...  I also see the green line blinking steadily during this, so there seems to be a weird software issue.

Note that I'd been trying with a fairly long startup delay, but that seems to have little effect.

jflyper, 2016-05-21:
So, your video includes the following sequence?

1. Power up with camera connected - sync problem.
2. B/W then color picture appears - no overlay
3. Camera disconnected
4. Overlay appears
5. Camera reconnected
6. Camera picture with overlay displayed

dustin, 2016-05-21:
Yes, sorry, that's correct.

It was getting late and I was done being awake. :)

Sometimes the overlay comes up initially, but rarely.

ShikOfTheRa, 2016-05-21:
maybe try:
//#define AUTOCAM                   // Disable if no screen display. Enables autodetect Camera type PAL/NTSC. Overrides GUI/OSD settings.
//#define AUTOCAMWAIT               // **UNTESTED** - Use with AUTOCAM - waits until camera is ready - i.e. if power up cameras after FC.

dustin, 2016-05-21:
It did seem to work somewhat reliably with AUTOCAM, but it would detect PAL
about half the time.  :/

jflyper, 2016-05-22:
Few more questions:

1. If you leave the camera connected, you will never get the picture?
2. Is this a new build?
3. If modding a previous build, tell us the area you've modded.
4. Tell us about your power supply configuration.

dustin, 2016-05-22:
Yeah, I never get overlay if it stays connected.

This is parts on a desk. Trying no-fc mode for a wing.

Power is ~3S worth coming out of my bench supply into a pololu switching
thing. Reading 5V off the input to the osd. Camera is taking power from the
same place (using pins from osd).

I added a cap across the inputs, but i assume it sags a bit at startup.
Adding delay does not help.  Sometimes it comes up, but often it doesn't.

jflyper, 2016-05-22:
I'm getting desperate ...

I would try switching to another micro minim if I have a spare, and/or a spare camera, before I look further into the software.

ShikOfTheRa, 2016-05-23:
OK, so what you say is interesting - with the AUTO stuff enabled, 50% of time it detects PAL, but always displays OSD?

To me that suggests the MAX chip is detecting the wrong camera type. Question is why? My guess would be the cam is out of spec. Would it be possible to try a different type?

Also which camera out of interest.

dustin, 2016-05-23:
I'm using a fatshark 600tvl cmos camera in this setup. Signal is configured
with a jumper, so is expect it to be stable.

ShikOfTheRa, 2016-05-23:
Weird how it keeps changing. Maybe this?

http://fpvlab.com/forums/showthread.php?24697-FAT-Shark-600tvl-Cam-NTSC-pal-issue

jflyper, 2016-05-28:
Ahh... I suddenly recalled that my FS600TVL had a slight sync issue, too.

It was not related to the "mineral deposit" mentioned in the above thread, but had to do something with the firmware. Don't remember the exact nature of the problem, since I decided not to use it at the time. Couple of month ago, I forgot all this and tried to use it in my desktop experiment, and had a sync issue again (of course).

ShikOfTheRa, 2016-07-21:
@dustin 
The latest in the repo has the best I think we can do for auto camera detect / initialization. 
It would be worth trying to see if it helps. Anything else I think we will have to put down to compatibility issues.
Have carried out some extensive testing with a number of cameras and I am getting consistent results now. There are definitely a number of cameras out there that have issues with max chip though.

ShikOfTheRa, 2016-08-04:
assume this is closed now. Any issues please re-open :)

closed 2016-08-04 by ShikOfTheRa