ShikOfTheRa/scarab-osd #244: Low Voltage Message not appearing

Hwurzburg, 2016-05-22:
Just loaded 1.6 on an OSD after running my version of 1.5 (with the autocells calc and mah status icon changes that are now in 1.6) and found that the low voltage warning message was not appearing on any layout...the battery voltage would flash, but no message in the alarm message area... 
I made a single simple change to the code and it started working by removing "&&!armedtimer" from this code in the Screen.ino display voltage sub shown below...this is the only change to make it work:
_#ifdef DISP_LOW_VOLTS_WARNING
  if (voltage<=voltageWarning&&!armedtimer)
    MAX7456_WriteString_P(lowvolts_text, getPosition(motorArmedPosition));
#endif_
this indicates that the timer is somehow not getting zero'd when it flashs the armed message or is being set to nonzero after this time....but I cant find it....any ideas? the only reason for the extended conditional is to prevent low voltage message from overwriting the brief armed message...

Hwurzburg, 2016-05-22:
Think I figured it out....if the layout doesn't include the ARMED/DISARMED/ALARM area, then the "armedtimer", which is set at 255 in global variable setup, never gets decremented in the displayArmed routine after arming since it immediately returns....this prevents the LOW VOLTAGE warning from being displayed....I will try setting the global initial value to 0.....

ShikOfTheRa, 2016-05-23:
Are you OK now?

Actually somewhere the armed status are became disabled in a couple of layouts. Sure I re-enabled but it keeps turning up to haunt me.... 
I do intend to revisit alarms / status display at some point. Try to improve.

Hwurzburg, 2016-05-23:
will get to test today or tomorrow....yeah, I only have every used three layouts (really only two in practice) and they both had the armed status, but one apparently dropped it in the 1.6 XML file...otherwise, I would have never found the issue...should be fine and have no other impact since the counter is always re-initialized to 20 upon arming in the main loop..
it did get me to learn how the layout editor worked though, which I had never explored before  :-)

Hwurzburg, 2016-05-23:
nope....turns out that the main loop sets the _armedtimer_ to 30 on any unarmed to armed transition for some reason...don't think its really necessary since its re-initialized before flashing the ARMED message to 20...so this is what it takes.....tested in many layouts with and without the armed field included in the layout...am not synched in github so someone would need to add the line in the screen.ino module...
_void displayArmed(void)
{
  if(!fieldIsVisible(motorArmedPosition))_
  **armedtimer=0;**
    _return;  ....._

Hwurzburg, 2016-05-23:
nope....the main loop always initializes armedtimer to 30....not sure why its there since displayArmed will initialize it anytime it gets flashed....so the code change below works...I have tested it in many layouts with and without the status line in the layout and it seems to work fine...I dont have github setup any more, so someone would have to make this simple change in Screen.ino and pull request..
_void displayArmed(void)
{
  if(!fieldIsVisible(motorArmedPosition))_**{
  armedtimer=0;**
    _return;_  **}**_......_

ShikOfTheRa, 2016-05-26:
Had a quick look at this.

The sole purpose of this is to flash "armed" for a count of 30 or 20

I'll update temporarily and revisit when looking at the alarms management. Opportunity to reduce...

ShikOfTheRa, 2016-05-26:
Done

closed 2016-05-26 by ShikOfTheRa