ShikOfTheRa/scarab-osd #245: Dronin Altitude incorrect + no Speed

cpusmith, 2016-05-23:
I'm not sure if I have MWOSD configured incorrectly or not but the Speed indicator always reads zero and the default Altitude reads -21 meters although it's pretty accurate in incrementing (-19 at 2 meters of height). I've added the GPS Altitude and it appears to start at 0 and increments but is not as accurate as the barometer Altitude which I'd prefer to use.

I'm using a Sparky 1.1 on a 250 frame with MWOSD 1.6 Master on a Micro MinimOSD and dRonin 2016-04-09.2 Release ("Tanto")

#234

tracernz, 2016-05-23:
We send the wrong altitude by the looks. It needs to be inverted in the case where it comes from PositionActual.Down (PositionActual uses the NED co-ordinate system). https://github.com/d-ronin/dRonin/blob/next/flight/Modules/UAVOMSPBridge/UAVOMSPBridge.c#L461

ShikOfTheRa, 2016-05-26:
Thanks tracernz. Doesn't sound like anything for MWOSD. 

Can someone advise if / when a patch is made available in dRonin

dustin, 2016-05-26:
You can just close this bug.  It's a dRonin thing.  I thought it had been
fixed over there, but not sure if it's come out in a release.

tracernz, 2016-06-02:
Issue raised to fix this.

ShikOfTheRa, 2016-06-02:
Re-opened to keep track

cpusmith, 2016-06-02:
Will you also be looking at the zero speed display as well?

tracernz, 2016-06-02:
Can do, on brief inspection looks like the only place we send a speed through is the raw GPS message.

tracernz, 2016-06-14:
@cpusmith Can you please test d-ronin/dRonin#1035 and see if it fixes your issues? Thanks in advance.

cpusmith, 2016-06-15:
Thanks.  Will do.

tracernz, 2016-06-24:
Confirmed fixed by @cpusmith. 👍

ShikOfTheRa, 2016-07-21:
Brill

closed 2016-07-21 by ShikOfTheRa