ShikOfTheRa/scarab-osd #246: Support S.Port telemetry

rtieland, 2016-05-25:
It would be nice if we could use S.Port for telemetry.

ShikOfTheRa, 2016-05-26:
I have no idea what you are requesting - not being an S port user... could you elaborate

barthalamu, 2016-06-10:
I'm not sure what he's thinking, but I've been pondering the idea of using the SmartPort output from my Taranis and send it to a local OSD connected to my goggles.

Since any data I care to recieve (volts, Amps, mAh, RSSI) is sent as telemetry anyway, it would free up a serial port, save some space and weight on the craft, and make the build a little cleaner. You could also use it to monitor your ground station battery and if your FPV feed is being broadcast, streamed, or otherwise sent elsewhere, you could send a clean uncluttered picture and only you would have the OSD info that you need.

I could also see eventual possibilities of getting other feedback from and/or control of your ground station, like VRX RSSI, diversity channel selection, lap times sent from a timing system, maybe even control of the VRX itself to change channels.

axyspg, 2016-07-02:
I think that it's similar to the idea I've posted on my blog. You can check it at:

http://someonedidthatalready.blogspot.com.es/2015/12/how-to-save-uart-port-on-your-msp.html

The idea is to free a serial port and send SmartPort telemetry using MSP data from MWOSD.

ShikOfTheRa, 2016-07-21:
Nice project ^^

I am going to close this as its not going to be developed. Sorry

closed 2016-07-21 by ShikOfTheRa