ShikOfTheRa/scarab-osd #248: RSSI showing 0%, with cleanflight and x4rsb

jeffwhu, 2016-05-29:
I am having much trouble setting up RSSI on 1.6 OSD, set "show RSSI" and  "use fc RSSI" and all that I'm getting is "0%" on the osd, also the frsky telemetry is working as I can get RSSI to display on taranis x9d plus, using uart2 on cleanflight set to MSP, shows battery voltage , horizon bar, timer, mode ok, callsign only flashes up briefly and RSSI stays on 0%.
Has me completely stuck.
Anybody know what I must do to get RSSI to display properly?
Regards
Jeff

eirism, 2016-05-29:
How are the FC, X4R and the OSD connected. See [this](https://github.com/cleanflight/cleanflight/blob/master/docs/Rssi.md) as to how to get RSSI in cleanflight.

jflyper, 2016-05-30:
Can you confirm the correct channel value in the CleanFlight Configurator? See the follow-ups  to your question in the fpvlab.com's thread for details.

jeffwhu, 2016-05-30:
Thanks for your input.
FC is sp f3 acro board configured to output msp from uart2, tx to rx and vice versa from uart2 to OSD which is powered from 5v distribution board ( reason for power is I thought 500mah for OSD would be too much for FC).
X4r sb connected to uart3 set as serial sbus on CFC.
Have open tx 2.1.7 on taranis x9d plus, getting RSSI on taranis but not on OSD, so, all points to setting in CFC, RSSI I have put on ch7 in CFC receiver tab, and set correct scale and weight set on input and mixer.
Tried telemetry in version on and off(not sure which is default on sp f3 -)
"erism", I have followed both the info on the two links you gave me previously. 
I am still baffled as to why the OSD is flashing 0% OR when applying inversion CFC solid on 50%, but still 0% on OSD.
Regards
Jeff

eirism, 2016-05-30:
What are the values of ch7 in CFC receiver tab, are they correct (I get about 1800 when close to the transmitter)?

jeffwhu, 2016-05-30:
I've just set rssi_ppm_nvert=on and get 100% on CFC I get rssi on set up tab 100% and not changing.
Not sure what you mean by the values of ch7?

eirism, 2016-05-30:
If you go to the reciever tab in CFC, you can see the received value. On your Taranis, what is the value of the RSSI channel on the Channel Monitor ([this](http://s29.photobucket.com/user/ridge-runner/media/RC_HELIS/TARANIS/20140426_TaranisLuaSetup_7of7_ChannelMonitor.png.html))

jeffwhu, 2016-05-30:
Ch7 on halfway on taranis, can't see a value on CFC

eirism, 2016-05-30:
[This](https://oscarliang.com/ctt/uploads/2015/07/RSSI-cleanflight-ppm-channel-receiver-tab-aux-showing-rssi.jpg?3186b4) is how it should look in the receiver tab in CFC. You RSSI channel on the Taranis should look something like [this (CH7)](https://oscarliang.com/ctt/uploads/2015/07/RSSI-taranis-ppm-channel-full-signal.jpg?3186b4)

jeffwhu, 2016-05-30:
[image]
This is what I'm getting

eirism, 2016-05-30:
Then I believe you haven't set up a RSSI channel correctly. Do you have a RSSI input, and then a RSSI mixer? The TX related part from [this](https://oscarliang.com/rssi-ppm-channel-taranis/)

jeffwhu, 2016-05-30:
Yes

eirism, 2016-05-30:
So the X4R are powered on, and CH7 on the Taranis is still -100? Can you take pictures of the INPUTS and MIXER pages of your model?

jeffwhu, 2016-05-30:
[image]
[image]

eirism, 2016-05-30:
Everything seems right there. What about the OUTPUTS page when you are highlighting CH7?

jeffwhu, 2016-05-30:
[image]
You mean this page?

eirism, 2016-05-30:
Yeah, and then "scroll" down to CH7 with the - button

jeffwhu, 2016-05-30:
[image]

jeffwhu, 2016-05-30:
[image]

eirism, 2016-05-30:
From what I can see, the RSSI input is good, but there is something wrong between the input and the output. Could you upload your model file?

jeffwhu, 2016-05-30:
How would I do that?

jeffwhu, 2016-05-30:
Got it from models but can't send via this page

eirism, 2016-05-30:
You can create a zip file containing the model file

jeffwhu, 2016-05-30:
[180.zip](https://github.com/ShikOfTheRa/scarab-osd/files/289499/180.zip)

jeffwhu, 2016-05-30:
That OK?

eirism, 2016-05-30:
Yes. The model you uploaded haven't got a mixer on CH7 (and isn't sending RSSI)

jeffwhu, 2016-05-30:
sent right model

eirism, 2016-05-30:
Try this model [180.zip](https://github.com/ShikOfTheRa/scarab-osd/files/289517/180.zip)

jeffwhu, 2016-05-30:
oh no, just wiped all models from taranis, and the file you sent does not work

eirism, 2016-05-30:
How does it not work? No RSSI? When using the simulator in Compantion TX CH7 behaves correctly.

jeffwhu, 2016-05-30:
just managed to wipe all other models from taranis

jeffwhu, 2016-05-30:
and still have o% on osd

eirism, 2016-05-30:
Then I don't know what's wrong. Good luck figuring it out

jeffwhu, 2016-05-31:
I don't know what was in the file you sent me.
But it has completely shafted my taranis, is there a way to completely reset taranis?

eirism, 2016-05-31:
The file I sent is a zip file containing a modified version of your model file (modified using OpenTx Companion 2.1.7. What did you do with the file? You might be able to use some of [this](http://openrcforums.com/forum/viewtopic.php?t=6016) to reset it.

jeffwhu, 2016-05-31:
[180.zip](https://github.com/ShikOfTheRa/scarab-osd/files/291366/180.zip)
its this on.... right?

jeffwhu, 2016-05-31:
put 2.1.8 now

eirism, 2016-05-31:
The link in [this](https://github.com/ShikOfTheRa/scarab-osd/issues/248#issuecomment-222708307) comment seems to be the zip with the original model file you sent. My OpenTX Companion (2.1.7) can read (and simulate) both the original model and my modified one. It might have been some mismatch between versions that broke things for you.

dustin, 2016-05-31:
As an aside:  There's nearly no value in putting RSSI on your OSD from
telemetry.  It's a bad idea, IMO.

frsky tx sends a data packet every 9ms for three packets in a row (i.e.
27ms) and then pauses to receive a telemetry packet.  So the rx will
transmit a telemetry packet roughly every 36ms.  Once that telemetry packet
makes it to the tx, the tx will then send it in the next outbound packet,
which is 18ms after the signal it's measuring.

So your very best case, you're displaying RSSI from 18ms - 45ms old.  (When
this becomes available for display is subject to protocol timings, so up to
another 5ms for s.bus + however frequently the OSD decides to poll, but
that's not as important here).

e.g.:

rx
rx  measured
tx  (send measurement back to tx) (+9ms)
rx receive measurement (+9ms)
rx  resused (+9ms)
rx  reused (but measured, +9ms)
tx (+9ms)
rx  delivered (+0)

"but why does this matter?"

There's this weird myth that RSSI is useful in this type of config because
the value slowly degrades over distance in some totally linear, predictable
fashion because that's how electromagnetic radiation propagation works.
But that's not how stuff you're flying works.

Your aircraft is moving around randomly directing the (probably damaged)
antennas in different directions fairly randomly.  In practice, you'll have
sudden packet loss caused by interference, falling into the antenna dead
zone, or even power problems.

Consider this case (I have many like this for a variety of radios):
https://gfycat.com/KeyEntireChital  (This is a radio speaking frsky d8
protocol, but whose firmware does direct channel injection of RSSI).

I don't have telemetry here, but if I did, what would I have seen?  I went
suddenly from 100% packet reception to 0%.  There was no packet received,
so nothing to report.  Perhaps receiving packets were dropped.  The
transmitter on the rx delivering the telemetry back to the tx is a whole
different set of problems, what if that 1 out of 36ms packets were
dropped?  I'd be displaying 100% RSSI and have no idea why my quad just
fell over.

In proper long range systems, this type of config is unlikely to even be an
option.  I've got a 1W tx and a 100mW (tx on the) RX.  I've long lost the
ability to hear the rx while still having great signal.  Of course, in
those systems, having the rx inject its RSSI on every packet is easy.

jeffwhu, 2016-05-31:
Thanks for your time even though this wasn't resolved

JoeB99, 2016-09-19:
Was this ever resolved? 
I have my taranis sending the RSSI back to the FC (naze32), Cleanflight Configurator shows the RSSI value correctly, along with the battery voltage on the Configurator Setup tab ->'Info' pane (right hand side).

The OSD shows the battery voltage from the FC correctly, but simply flashes 0% for RSSI.

Cleanflight v1.13
CF Configurator v1.2.3
OSD v1.6.2 DEV

jflyper, 2016-09-19:
Your problem is different.

Did you turn on `USE FC RSSI` in the MW_OSD_GUI?

JoeB99, 2016-09-19:
Yes, Use FC RSSI is enabled in the GUI.

jflyper, 2016-09-19:
I just confirmed it working.

Clmeanflight v1.13 (+ iNav 1.2)
CF Configurator v1.2.3 (+iNav Configurator 1.0.2)
OSD v1.6.2 DEV

Stock Config.h (as of v1.6.2.0)

jflyper, 2016-09-19:
It should work. Why not do a fresh start.

(1) Open MW_OSD_GUI and take note of important settings like calibration values.
(2) Download a fresh copy of R1.6.2.0.
(3) Compile & upload
(4) Connect with MW_OSD_GUI, hit SETTINGS-DEFAULT (wait until restart complete)
(5) Set RSSI-Display RSSI & Use FC RSSI
(6) Hit SETTINGS-WRITE
(7) Disconnect MW_OSD_GUI
(8) Switch to FC connection and test.

jeffwhu, 2016-09-19:
where is R1.6.2.0?

JoeB99, 2016-09-19:
Yes, that works, but why does it not work on my custom screen layout?
[j1.osd.txt](https://github.com/ShikOfTheRa/scarab-osd/files/480237/j1.osd.txt)

 (attached, remove .txt)

jflyper, 2016-09-19:
Where did you get the j1.osd from?

JoeB99, 2016-09-19:
That't the output of DISK -> "SAVE"

1.6.2 is here: https://github.com/ShikOfTheRa/scarab-osd (close or download -> Download zip)

jflyper, 2016-09-19:
Oh, okay :)

dustin, 2016-09-19:
I've used this config on several different flight control systems with
mwosd and not had an issue.  Your radio doesn't do it correctly, so you
can't possibly get useful information (as I mentioned earlier), so faking
it is a very bad idea, but if you fake it, but it definitely works.

(But seriously, don't fake it.  Bad information is worse than no
information)

It'd be much more useful to connect the RSSI pad from your radio to the
flight controller with an RC filter and using the same config config you're
using here, or hooking it up to an ADC on your osd and doing it there.

dustin, 2016-09-19:
> I'm concentrating on FC supplied RSSI not being displayed correctly (the
> feature should work regardless of the source of the RSSI).
>
> @JoeB99 <https://github.com/JoeB99>
> Yeah, I confirmed that with your layout, it displays zero regardless of
> the FC supplied value.
> Let me look into this.

The relevant parts of that config:

     <entry key="Display RSSI">1.0</entry>
     <entry key="RSSI Alarm">50.0</entry>
     <entry key="RSSI min">0.0</entry>
     <entry key="RSSI max">0.0</entry>
     <entry key="Use FC RSSI">1.0</entry>

rssi min and max are both 0 in that config.

dustin, 2016-09-19:
> @dustin <https://github.com/dustin>
> You rock!

I just adjust the values on the OSD until they match what I expect.  I'm
using 150 with some of my radios, and like, 7 with others.

jflyper, 2016-09-19:
For receiver channel PPM/PWM RSSI, min-max should be 0-1023.

For analog (ADC) based RSSI, it depends on how RSSI is translated into RSSI ADC input voltage.

jflyper, 2016-09-26:
@jeffwhu 
Sorry I overlooked your question

> where is R1.6.2.0?

You can get it from [Releases](https://github.com/ShikOfTheRa/scarab-osd/releases) page.
However, I believe it doesn't solve your problem if you are still at where you were on June 1.
JoeB99 already had a correct reading in RSSI in the configurator.

jeffwhu, 2016-10-06:
Will close after I've written this, to be honest I've given up on it, still not working, it's not an absolute" must have" , but would've been good to have as the reception I get where I fly is great one and diabolical the next, still not worked out why, the only thing that is near by is a fire station, don't know if that would affect my signal strength, does affect 5.8ghz video signal as well.
Thanks for your input
Regards
Jeff

closed 2016-10-06 by jeffwhu

ShikOfTheRa, 2016-10-06:
@jeffwhu I think probably get better support in one of the FPVlab threads.

Also check the MWOSD FAQ. Thes is a link to instructions. Don't use myself but it may help....

FAQ 24 How do I use RSSI with FRSKY (D4R-II)

jeffwhu, 2016-10-06:
Hi

Thanks for that, I can get the rssi with d4r-ii but not with the x4r sb,  it thanks for all the advice

Regards

Jeff