ShikOfTheRa/scarab-osd #250: Add VEL pid coefficients for altitude hold.

rb1205, 2016-05-31:
The PID menu page makes possible to modify the coefficients for the ALT pid but not for the VEL pid, which is used in cleanflight's altitude hold function to regulate vertical velocity (but I assume is the same for any multiwii-based firmware). Being able to edit those coefficients is required while tuning the altitude hold function.

GlobalVariables.h seems to define all the required screen positions (VELT through VELD), however the field name is not definied in the configMsg1* area, nor in menu_pid.

jflyper, 2016-05-31:
Hmmm. I took a look at the code.
It displays what FC told OSD to display; it isn't that easy as adding an extra line for VEL pids.

rb1205, 2016-05-31:
I'm not that well versed in C, however in [Serial.ino line 409](https://github.com/ShikOfTheRa/scarab-osd/blob/master/MW_OSD/Serial.ino#L409) the code

` if((pn_index != 5) && (pn_index != 6) && (pn_index <= 8)) // 5, 6 and >8 are skipped`

Skips some pid index by design. I don't know the details of the serial format so I can't say for sure, but chances are VEL is just being skipped here.

jflyper, 2016-05-31:
@rb1205 Ah, yes. That is part of the code. There are some more in Screen.ino also.

jflyper, 2016-05-31:
@rb1205 
I think I have a working code, but it needs testing.
[https://github.com/jflyper/scarab-osd/tree/pid-menu-with-vel](url)

I also have to find out the relationship of this option (`MENU_PID_VEL`) with other configuration options; currently, the option is turned on only for CLEANFLIGHT>1.9

rb1205, 2016-05-31:
sorry, I tried your branch but I failed to compile it :(

There's a typo on line 1132 of Screen.ino (#if -> #ifdef) which I fixed, but I keep getting a strange "MW_OSD.ino:143:1: error: expected unqualified-id before ‘else’". I also tried to cherry-pick the last commit while on the master branch, with no success.

Can you fix it, please? I'd be more than happy to test it.

jflyper, 2016-05-31:
Hmm... Can you try again?

jflyper, 2016-06-01:
Oh, wait. It's still there with freshly downloaded zipped fork... Let me look into this.

jflyper, 2016-06-01:
Okay, you can try again now.

P.S. The trick:
I found out that IDE 1.0.x does not allow unbalanced braces created with preprocessor directives,  e.g.,
```
#ifdef SOMECASE
    while (expression1) {
#else 
    while (expression2) {
#endif 
        // while body
    }
```
which is perfectly legal AFAIK, produces an error !!!

tracernz, 2016-06-01:
Preprocessor should replace that section with one or the other before the compiler ever sees the source. I guess Arduino decided to ignore that. What happens if you do something like this (I'm not suggesting to do it in production code, just for interest):
```c
#ifdef SOMECASE
    while (expression1) {
#else 
    while (expression2) {
#endif 
        // while body
#ifdef SOMECASE
    }
#else
    }
#endif
```

jflyper, 2016-06-01:
@tracernz 
It's off topic, but your snippets both fail to compile.
... well, personally, I found another reason to leave 1.0.x behind 🍺

rb1205, 2016-06-01:
@jflyper 
Works like a charm :D
It parses correctly the values from the FC, it shows them in the pid menu at the last line under "VEL" label, it sets and saves them as expected. I'll do some additional test later, but looks very good as it is.

The only remark I have is that the ALT pid is in the middle of the menu, while VEL is at the bottom after LEVEL and GPS. I know that's just the order they came from the FC, but it's a little confusing if you don't know what the VEL pid is about. The better solution would be to move it under ALT pid like it is in the configuration tool. If that's too hard/ugly to implement, changing the label from VEL to something like Z_VEL or ALT_VEL could be an easy and effective solution.

jflyper, 2016-06-01:
@rb1205
Glad to hear it worked.

Yeah, I felt a bit uncomfortable looking at the order of items, too, but thought it would be better to produce a working code first. Let me see if I can do something with in the code size constraint.

ShikOfTheRa, 2016-06-01:
@jflyper 
Maybe its time for a different approach to PID and RC tuning.
The constraint has been fitting everything on an NTSC window. Maybe we should provide access to all settings, but provide an 8 line "window" into them using an index pointer. So can change anything. Also use MSP PIDNAMES which would help with the different variants.

Good use for all that memory you freed up !!

jflyper, 2016-06-02:
@rb1205 
I changed the label to "Z_VEL"; tried few things, but couldn't come up with clean&easy&compact way of reordering items.

@ShikOfTheRa 
There's something going on here: cleanflight/cleanflight#2014

ShikOfTheRa, 2016-06-02:
This would be fantastic. We have to support...

jflyper, 2016-06-05:
@rb1205 
VEL support is now merged into the master.
I had to do some rebasing, so it is nice if you can test the current master.
Once you are done, you can close this issue.

rb1205, 2016-06-06:
Sure, will check this evening.

Thanks!

rb1205, 2016-06-06:
Just tested the main branch, and as far as I can see everything seems to be the same as in your previous branch! I'm closing the issue, thank you very much!

closed 2016-06-06 by rb1205

jflyper, 2016-06-07:
Cool! Thanks!

ShikOfTheRa, 2016-06-07:
Cool +1