ShikOfTheRa/scarab-osd #254: mw osd

alshemali320, 2016-06-03:
hello
  am seing windows 10 ,, i am try to install my mw osd all setup went ok , except when i run the program ,, the reading osd hung at 36% and wont be complete , i redo all setting when no help ..
 best regard

jflyper, 2016-06-05:
Need more information on the OSD hardware, MWOSD release, nature of USB driver you are using, instruction page if you followed one, etc.

BTW, did you successfully flashed the OSD with Arduino IDE?

jflyper, 2016-06-05:
Oh, and it is better for you to ask this kind of question in the fpvlab's MWOSD thread. There are much more people around there.
[http://fpvlab.com/forums/showthread.php?34250-MWOSD-for-MULTIWII-NAZE32-BASEFLIGHT-HARIKIRI](url)

closed 2016-07-21 by ShikOfTheRa