ShikOfTheRa/scarab-osd #258: Signal type on intro is somewhat misleading.

jflyper, 2016-06-11:
Currently, the signal type on the intro screen tells you if AUTOCAM is on, and the actual setting of the MAX7456 (Settings[S_VIDEOSIGNALTYPE]).

> AUTO-NTSC: AUTOCAM is on, MAX is set to NTSC
AUTO-PAL: AUTOCAM is on, MAX is set to PAL
NTSC: MAX is set to NTSC
PAL: MAX is set to PAL

The problem is, the AUTO-NTSC and AUTO-PAL does not always mean the standards are actually detected. Here is the code.
```
  if ((B00000001 & srdata) == B00000001){     //PAL
      Settings[S_VIDEOSIGNALTYPE]=1;
      autocam_detected = 1;
  }
  else if((B00000010 & srdata) == B00000010){ //NTSC
      Settings[S_VIDEOSIGNALTYPE]=0;
      autocam_detected = 1;
  }
 // If neither PAL or NTSC is detected, then the original Settings[S_VIDEOSIGNALTYPE] is preserved. 
```
For example, when the S_VIDEOSIGNALTYPE is PAL, even if MAX couldn't detect the signal, the intro screen  will display AUTO-PAL. The AUTO-PAL is also displayed when the MAX actually detected the PAL regardless of the S_VIDEOSIGNALTYPE.

This is confusing. The message should help the user to identify:

```
AUTOCAM: active or not
Signal detected: PAL, NTSC or **NONE**
S_VIDEOSIGNALTYPE: PAL or NTSC
```

So, the message would be something like:
AUTO-PAL-PAL
AUTO-NTSC-NTSC
AUTO-NONE-PAL
AUTO-NONE-NTSC
PRESET-PAL
PRESET-NTSC

ShikOfTheRa, 2016-06-12:
Yes - am also aware of this. 

I have amended since as a result of testing below but not uploaded. Gone with following KISS
PAL
NTSC
NOT DETECTED

Reason have removed AUTO is assumption is user knows as they have made a conscious decision to select AUTOCAM. And saves a few bytes Might just put it back in though.

Testing:
Added this in because I'm doing some testing with various different cams at the moment to determine if can use autocam/autocamwait enabled as default. Its a bit time consuming actualy. Different copters/ planes / cams / connectors / voltages!. Am getting some inconsistent results although it may well be the cams.

Whatever do I don't want to use much code on something that is displayed for just a couple of seconds!

ShikOfTheRa, 2016-06-13:
PR committed this morning that includes amendment above.

From my testing it does appear that most cams give 100% compatibility. 
The nextchip 2090 based ones seem a lost cause. Consistent field issue.
Others seem a little unpredictable. Not sure but it almost seems like they initialize with NTSC or something that looks like it then change to PAL very quickly. Maybe it is this that is confusing max chip. Not sure if a reset max after delay is working once chip has detected a signal? Or maybe this is how the chip functions and it has to loose sync before it redetermines signaltype.

jflyper, 2016-06-13:
Cams initializes as NTSC and quickly change to PAL or vice versa is very likely to happen with a careless firmware programming.

It is going to be extremely useful to fully understand how max behaves against input signal transitions, in terms of:
(a) external signal type, NONE, NTSC and PAL
(b) timing of the signal type changes
(c) configured video standard v.s. external signal detection (i.e. will max still detect incoming signal type if video standard is configured)
(d) active and passive sync generation.

ShikOfTheRa, 2016-06-13:
To test  am going to have to have to connect up 2 cams with a switch and flick between them. 
Even then what can we do? I wonder if adding a check of the signaltype detected into the maxstalldetec routine. If it !oldcheck then reset max chip. It worked well for the power issues.

jflyper, 2016-06-14:
Yeah, adding LOS and signal type check in the stall detection is what I had in mind. And, for doing so, we have to know the max's behavior some what more in detail, I believe. The knowledge would be great for trouble shooting, too.

jflyper, 2016-06-16:
@ShikOfTheRa 
I'm closing this one for now. Let's wait and see how people (and their cams) react to the hack.

closed 2016-06-16 by jflyper