ShikOfTheRa/scarab-osd #259: problems with MWOSD R1.6 in linux and 4K (non-HiDPI) screen

sl1pkn07, 2016-06-12:
[image]

the window is cut in bottom, and the text is too small

[image]

tested with JDK 1.7, JDK 1.8 and JDK 1.9 (all Oracle)

the screen is a 4K panel but non-HiDPI compilant (157x161dpi)

greetings

ShikOfTheRa, 2016-06-13:
Sorry - can't help with this one... no idea

sl1pkn07, 2016-06-13:
i can test this?

https://github.com/ShikOfTheRa/scarab-osd/issues/198#issuecomment-197829535

greetings

ShikOfTheRa, 2016-06-14:
#198 is still being developed

ShikOfTheRa, 2016-07-21:
Am going to have to close this as its not going to be resolved.

I can only suggest using a virtual environment if needed. Or wait until alternative GUI is available. Not ETA for that...

closed 2016-07-21 by ShikOfTheRa