ShikOfTheRa/scarab-osd #268: MW_OSD_GUI/application.macosx gone?

jflyper, 2016-06-22:
Looks like it was removed as a part of #261?

ShikOfTheRa, 2016-06-24:
Hi - no change to layout, but unfortunately there were some release out there with bad xml file causing lockups when functions enabled. 
Wanted to force this for the last point release to make sure they didnt use an exisiting GUI. 
If it helps, I'm happy to restore the gui and downrate the ver. Its really  not an issue :)

jflyper, 2016-06-26:
Is there any problem with MWOSDVER==13 and application.macosx ???

ShikOfTheRa, 2016-06-27:
No - I just cant create the application

jflyper, 2016-06-27:
Any time frame for the MacOSX GUI?

I can live with MWOSDVER reverted to 12 in my local repo, but I'd suspect the majority of the mac users can't.

LaserLock, 2016-06-27:
you can built it yourself with https://github.com/ShikOfTheRa/scarab-osd/tree/master/OTHER/MW_OSD_GUI_SOURCE/MW_OSD_GUI

ShikOfTheRa, 2016-08-04:
reverted :)

closed 2016-08-04 by ShikOfTheRa