ShikOfTheRa/scarab-osd #270: Flying time

domsmod, 2016-06-24:
Hello,

I have updated my micro osd with version 1.6 and now it lasts there no more fly time in option in the layout editor !!!

I do not want to post the time of startup but only when I arm my multirotor.

Thanks you for your help

jflyper, 2016-06-25:
From what I can read from the code is that there is only one timer (named "Timer" in GUI) on the screen,  and it shows "on time" when the vehicle is not armed, and switches to "fly time" when the vehicle is armed.

jflyper, 2016-06-27:
Yes.
The above mentioned behavior is confirmed.

domsmod, 2016-06-27:
Precisely, when I arm total time is to post, out I want that it has zero has each armament, if not I can have 7 min flight to post and they are not good

And "reset after arm" does not put has zero

jflyper, 2016-06-28:
The "reset after arm" only works for the statistics screen.

So, you want to ...
(1) Always display the "Fly time".
(2) Resetting "Fly time" to zero on each new arm.
(3) Both of above.

Which?

BTW, what version of the firmware you were using previously (before R1.6)?

closed 2016-07-26 by ShikOfTheRa