ShikOfTheRa/scarab-osd #273: add support for BetaFlight custom MSP commands

csurf, 2016-07-01:
Betaflight 2.8.0 & above has changed the way rates/expo works, and has added rc_rate_yaw to control center stick sensitivity for yaw. This value is controlled via MSP_SPECIAL_PARAMETERS. There are a few other custom MSP commands for BF that might be useful to change. Perhaps there could be an optional BF-specific menu that lists some of the more important BF parameters?

ShikOfTheRa, 2016-07-05:
Hard to keep up with all changes across all controllers... ! Is it just an extra menu / MSP or does the existing menu change also? 

Need to support as its so popular

csurf, 2016-07-08:
would probably be easier to just add an extra menu, as it would avoid having to mess with existing code, and it could be easily turned off via config file for non-betaflight users. Your thoughts?

FYI, here's a link to the existing betaflight MSP commands...
https://github.com/betaflight/betaflight/blob/master/src/main/io/serial_msp.h#L196

waltr1, 2016-07-08:
Just a vote to have a update to MWOSD as I use this extensively for tuning.
Adding another #define for higher versions of ßF like done for CF would be perfect.

tracernz, 2016-07-08:
The joys of a (non-standard) standard protocol. Why not just redesign the protocol once to cater for all of the new needs and be done with it?

csurf, 2016-07-09:
@tracernz 
would be nice, but it's way beyond the scope of MWOSD. That would need to be done at the CF/BF level, as they'd have to come up with a whole new protocol that's reflexive/self-defining, so that any future changes to the protocol definitions can easily be pushed out to the clients (OSD, config tools, etc.).

For now, we're stuck dealing with MSP and having to update the protocolo definitions on the consumer end every time the protocol is updated/changed.

dustin, 2016-07-09:
@csurf, if you're extending a protocol frequently, it's good to take a step
back and figure out if there's a better way to solve the class of problems
you're working on.  E.g., how many special protocol extensions would one
need to update settings?

@waltr1, ß is way different from β -- the former is closer to "ssF"

tracernz, 2016-07-09:
> if you're extending a protocol frequently, it's good to take a step
back and figure out if there's a better way to solve the class of problems
you're working on.  E.g., how many special protocol extensions would one
need to update settings?

Exactly. 👍

waltr1, 2016-07-09:
ß is way different from β
Yes but I never found the ALT key code for the β. Could you tell me how to type the beta character?

Since MSP is quite old now maybe it is time for a new protocol.

csurf, 2016-07-09:
@tracernz, @dustin 
that's all great, but, again, outside the scope of what's going on here...This is the code repo for the MWOSD, and not for betaflight/cleanflight. If you want MSP to go away & be replaced with something else, that needs to happen at the CF/ßF level, and not here. 

MWOSD is simply a client that relies on the existing MSP protocol in order to communicate with the FC (same goes for the EZ-GUI phone app and the 'CF/ßF configurator' chrome apps; they function via MSP). In order to get MWOSD to work with the latest iterations of ßF, the newest ßF MSP config commands need to be manually added to the MWOSD code. No, it's not ideal, but that's currently how it works across all config tools. The priority right now is to get it working, versus having a long discussion about overhauling the serial protocol (which isn't something that can be done here anyway).

dustin, 2016-07-09:
If a system such as "ss Flight" is making changes you want available in the
OSD, it's not necessarily the OSD software's responsibility to follow each
individual system's ad-hoc extensions.  Perhaps we should work together to
make a better system rather than shoving all the complexity downhill.

It's already painful enough for the users.  MWOSD has a ridiculous amount
of code to work around tiny differences from one system to another.  Things
as simple as the CORRECT_MSP_* definitions to do weird things, different
number interpretations, etc...

tracernz, 2016-07-09:
@csurf Moving the burden onto 3rd parties, as betaflight seems to do, isn't really the right way to do things IMO.

csurf, 2016-07-09:
@dustin @tracernz 
again, that's all fine&dandy, but it's not going to happen here. 

Yes, it sucks, but the onus is on the CF/ßF devs to come up with a new protocol, and *not* on shikra to try to impose something on the *Flight community. 

Perhaps it would make more sense for you guys to go and open up a discussion/issue on the ßF repo and push for the idea of having some kind of OSD protocol that automatically provides all the necessary info over serial without the OSD firmware itself being aware of each&every configuration parameter. dunno, just a thought.

In the meantime, as hacked & ineffiicient as it may seem, we simply need an additional menu group / MSP set added to MWOSD that can handle the essential ßF-related config parameters.

ShikOfTheRa, 2016-07-21:
Happy to add in additional for this.

It would be really nice to come up with a generic MSP config request for sure. If someone liek to work with me to develop I'd be happy to support. I think the long term future is embedded OSD though.

ShikOfTheRa, 2016-07-21:
These seem to be the changes. Which are most likely to be used? Probably cant support all of them... 
case MSP_SET_PID_ADVANCED_CONFIG :
case MSP_SET_FILTER_CONFIG :
case MSP_SET_ADVANCED_TUNING:
case MSP_SET_TEMPORARY_COMMANDS:

csurf, 2016-07-22:
yaw rate is crucial, which I believe is part of SPECIAL_PARAMETERS, which sucks because it's the only useful parameter in that MSP set.

FILTER_CONFIG might be useful in order to adjust the gyro & d-term filters.
I don't personally think ADVANCED_TUNING is very useful during flight, but others would need to voice their opinions.

I don't see SET_TEMPORARY_COMMANDS. Maybe you meant SPECIAL_PARAMETERS?

ShikOfTheRa, 2016-07-22:
OK - lets wee what others chime in with. re "special param" - I haven't looked at the latest - think it was the first cut so prob changed

Tomcatt, 2016-08-24:
Can you do this for all for BF, CF, and RF?  If I could code, I'd help you out.

proggod, 2016-08-24:
Would be happy to give some raceflight support too

Tomcatt, 2016-08-24:
Oh yes, hook them up.

proggod, 2016-08-25:
Join our slack http://raceflight.net/slack if you ever want to discuss rf + osd stuff, i'm not the best at keeping an eye on all these gits

jflyper, 2016-09-01:
I was expecting [Configuration menus](https://github.com/cleanflight/cleanflight/pull/2014) to become  more concrete sooner, but dominic seems to be caught with something else. A side from the design of the current PR code, the idea itself look theoretically correct.

Yes, we can add variant specific menu pages as pragmatic solution, but it's a pain in a bu** :censored: My software engineering soul is telling me to define a menu description language and a preprocessor to generate a menu code...

ShikOfTheRa, 2016-09-13:
@csurf  am doing this now. Which of the following params are the ones needed?

serialize16(currentProfile->pidProfile.rollPitchItermIgnoreRate);
        serialize16(currentProfile->pidProfile.yawItermIgnoreRate);
        serialize16(currentProfile->pidProfile.yaw_p_limit);
        serialize8(currentProfile->pidProfile.deltaMethod);
        serialize8(currentProfile->pidProfile.vbatPidCompensation);
        serialize8(currentProfile->pidProfile.ptermSRateWeight);
        serialize8(currentProfile->pidProfile.dtermSetpointWeight);
        serialize8(0); // reserved
        serialize8(0); // reserved
        serialize8(currentProfile->pidProfile.itermThrottleGain);
        serialize16(currentProfile->pidProfile.rateAccelLimit);
        serialize16(currentProfile->pidProfile.yawRateAccelLimit);

ShikOfTheRa, 2016-10-06:
Any feedback? If not I'll close this and re-open

The serial code is in there. Just need to decide which options to display on menu

rav-rav, 2016-10-10:
I think the initial request was to add superexpo support for roll/pitch/yaw in MSP_RC_TUNING:
`for (i = 0 ; i < 3; i++) {
    serialize8(currentControlRateProfile->rates[i]); // R,P,Y see flight_dynamics_index_t
}`

Regarding MSP_PID_ADVANCED I personally like to see _setpointRelaxRatio_ and _dtermSetpointWeight_ as they really change the feel of the quad. The other settings might be useful too, but hard to tune without seeing blackbox.

ShikOfTheRa, 2016-10-11:
I'm confused over what is required from the intial post tbh.. - what you referenced is already in the OSD menu.
Just looked through MSP_RC_TUNING and the only value sent via MSP that is not in OSD menu is rcYawExpo8. Would that be it??

Thanks for feedback on the RC_Tuning

rav-rav, 2016-10-11:
Oh, you are right, it's all there except rcYawRate8
rollRate/pitchRate/yawRate just works different now, but MSP has remained the same.

ShikOfTheRa, 2016-10-11:
Thanks for checking

is rcYawExpo8 significant enough to be added? IS it likely to be used more than the MSP_RC_TUNING options?

rav-rav, 2016-10-11:
rcRate8 and rcExpo8 is only used for roll/pitch
rcYawExpo8 and rcYawRate8 is only used for yaw

So in order to have full control over yaw, it should be included. These options are in MSP_RC_TUNING, I guess you meant MSP_PID_ADVANCED in your last question? MSP_RC_TUNING is probably used more often than MSP_PID_ADVANCED .

ShikOfTheRa, 2016-10-11:
yeah typo...
Will leave open for a while to see if anyone chimes in. The less I can include in menus the more it saves memory. Only want the essentials. At least until there is a generic configuration method for all items

jflyper, 2016-10-11:
I think I can make the bfosd's configuration menu start using external mwosd before it starts snowing here.

ShikOfTheRa, 2016-10-11:
bf has that now?cool

jflyper, 2016-10-11:
It only works for fc integrated osd now, but I believe it's not too hard to send draw(x, y, string) kind of request over the msp link insead of directly calling max7456drawstring().

ShikOfTheRa, 2016-10-11:
I guess something like......

OSD requests config item number. number  being 0-x or whatever. 16 bit value

FC sends:
descriptor length
display format - decimal position
min
max (if max=1 then display on/off instead of a value)
value (32 bit interpretation of value)
descriptor - max 20 chars

OSD save... writes back
config number 16 bit value
data 32 bit value

jflyper, 2016-10-11:
It will be more ... say, naive.

The menu system actually has it's own (virtual) screen and knows how to layout items. OSD subsystem will be sent with changes to the screen by display(y, x, string) type API.
The OSD subsystem will draw whatever it is told to draw, without any knowledge of what they really mean.

ShikOfTheRa, 2016-10-12:
Ah gotcha. That is even better. And sounds easy module to add.

jflyper, 2016-10-13:
I already have most of the code ;)

I even implemented INVERTED character support (which turned out to be a waste for this extension!!!)

ShikOfTheRa, 2016-10-13:
is this compatible with beta/clean/race ? Or are tehy all doing different thinks now

jflyper, 2016-10-14:
Our first implementation of the canvas mode will be compatible with BF, because only BF has a deployed configuration menu system (even though it's part of the integrated OSD atm). If the canvas approach ever get accepted by users, then it will (naturally) be compatible with more systems :)

ShikOfTheRa, 2016-10-14:
OK. Sounds good. Hopefully can get others to support

jflyper, 2016-10-15:
Okay, the canvas mode hack is working.

Now cutting into the BFOSD code.

jflyper, 2016-10-16:
OK. It works 🍺 

[image]

ShikOfTheRa, 2016-10-17:
Not enough Beers !!! Fantastico
:beer: :beer: :beer:

jflyper, 2016-11-25:
Yeap. It will be released as a part of the Betaflight v3.1.

The project has been extremely successful, and has gone way beyond what I was expecting to do; the Betaflight CMS (Configuration Menu System) is now display device independent, and runs on integrated OSDs, external OSDs (like MWOSD of course), I2C OLED displays. These display devices now all have displayPort abstraction which is (x,y) addressable screen, and CMS as well as OSD is now a client of the displayPort devices. Pretty nice. I believe we can write more clients, such as warning systems and such.

ShikOfTheRa, 2016-11-25:
Excellent. I just wish we didn't have to have all the different flavours of xxxxFLIGHT out there... 

So with MWOSD, betaflight v3.x is defined and includes canvas mode - am I right in thinking it autodetects - so can run 3.1 with canvas mode or 3.0 without just fine?

jflyper, 2016-11-26:
> Excellent. I just wish we didn't have to have all the different flavours of xxxxFLIGHT out there...

Maintenance of menu contents that manipulates FC firmware parameters now became the responsibility of FC firmware devs and maintainers, at least for Betaflight for now. We can kindly reroute "Support for SecretFlight Ultra-Rate adjustment" kind of request back to FC devs and maintainers now ;)

> So with MWOSD, betaflight v3.x is defined and includes canvas mode - am I right in thinking it autodetects - so can run 3.1 with canvas mode or 3.0 without just fine?

Even with the CANVAS enabled, MWOSD runs just fine with 3.0. It autodects ... yeah, you can say that, it autodects 3.1 when it receives the MSP_CANVAS (will probably be renamed to MSP_DISPLAYPORT to be coherent with BF) message.

Actually, the real problem is that whether or not we decide to drop menu code for betaflight parameters (or make them optional) in trade for some more flash space. Dropping entire FC related menu may be considered in a long range.

ShikOfTheRa, 2017-01-10:
jflyper's cms is in and working to cover all  so think this can be closed.

Also added yaw to the advanced page today.

closed 2017-01-10 by ShikOfTheRa

xymlinux, 2020-09-22:
> 天真地……会更多。
> 
> 菜单系统是整个系统中唯一了解菜单结构和项目的模块。它具有自己的（虚拟）屏幕，并且知道如何布置项目。OSD子系统将通过display（y，x，string）类型的API随屏幕更改而发送到屏幕，并且它们会根据告诉绘制的内容进行绘制，而不会真正知道它们的含义。我将其称为CANVAS模式。

how to enter canvas mode?

xymlinux, 2020-09-22:
> > Excellent. I just wish we didn't have to have all the different flavours of xxxxFLIGHT out there...
> 
> Maintenance of menu contents that manipulates FC firmware parameters now became the responsibility of FC firmware devs and maintainers, at least for Betaflight for now. And we can probably kindly reroute "Support for SecretFlight Ultra-Rate adjustment" kind of request back to FC devs and maintainers ;)
> 
> > So with MWOSD, betaflight v3.x is defined and includes canvas mode - am I right in thinking it autodetects - so can run 3.1 with canvas mode or 3.0 without just fine?
> 
> Even with the CANVAS enabled, MWOSD runs just fine with 3.0. It autodects ... yeah, you can say that, it autodects 3.1 when it receives the MSP_CANVAS (will probably be renamed to MSP_DISPLAYPORT to be coherent with BF) message.
> 
> Actually, the real problem is that whether or not we decide to drop menu code for betaflight parameters (or make them optional) in trade for some more flash space. Dropping entire FC related menu may be considered in a long range.

how to enter canvas mode?

ShikOfTheRa, 2020-10-01:
Hello - MWOSD it enters automatically when it detects canvas command from Flight controller.
example would be throttlemid, yaw left and pitch full to get into OSD menu

Also there is this. It is probably not what you want - the FC runs permanently in canvas mode. All screen layouts etc are configured in teh FC. Fonts installed need to be from the FC. It is unusual... 

https://github.com/ShikOfTheRa/scarab-osd/wiki/CANVASOSD