ShikOfTheRa/scarab-osd #274: RSSI disappears after some minutes

alexFab, 2016-07-03:
I use mwosd as standalone, just for displaying RSSI and Voltage. It works fine, but sometimes the RSSI disappears after some Minutes. It doesn't show 0%, it shows nothing. Sometimes the it comes back after a while. 
At first i thought it is a wiring issue, but than i realize in the case of an wiring issue the rssi would show 0%.

jflyper, 2016-07-04:
Which release are you using?
How long does it usually take to develop the symptom?
Does it develop on the ground also?
Does the voltage remains displayed when RSSI disappears?

alexFab, 2016-07-04:
using 1.6. Hard to how long it takes, i would say some minutes. I am quite sure i never seen this on the ground. The Voltage stays on the OSD whne RSSI disappears.

jflyper, 2016-07-04:
Can you attach Config.h, Def.h you are using, and what ever you have changed from R1.6 distribution?

ShikOfTheRa, 2016-07-05:
Weird - can't think of anything that would cause that.

jflyper, 2016-07-05:
@ShikOfTheRa I was thinking of something over writing the RSSI for a period of time.

ShikOfTheRa, 2016-07-05:
Possibly... maybe callsign full of blank space. I also wondered that but I'd expect that to be quite short unless setting exchanged from default.

alexFab, 2016-07-05:
sure, here are the two files i touched.

[Config.h.txt](https://github.com/ShikOfTheRa/scarab-osd/files/348362/Config.h.txt)
[Def.h.txt](https://github.com/ShikOfTheRa/scarab-osd/files/348361/Def.h.txt)

jflyper, 2016-07-06:
@alexFab Config and Def look decent... 
How about the call sign discussion above? Any thing that rings your bell?
How often do you see this symptom develop?
Can you reproduce it on the ground?

alexFab, 2016-07-06:
nothing done regarding the call sign. Never saw it on the ground. It happens nearly every flight.

jflyper, 2016-07-06:
How long does your flight usually last?
Can you reproduce the problem on the ground if you keep it powered for the duration of your average flight?

alexFab, 2016-07-06:
i will try it maybe in some days.

ShikOfTheRa, 2016-07-21:
Any update - also is it possible to see a video of it happening

closed 2016-07-26 by ShikOfTheRa