ShikOfTheRa/scarab-osd #276: The new MAX initialization code

jflyper, 2016-07-08:
Direct port manipulation was one thing I was avoiding; does this bring substantial savings?

```
/********************  OSD HARDWARE rule definitions  *********************/
#ifdef RUSHDUINO // Example for Arduino guys                     
    # define DATAOUT          11 // MOSI
    # define DATAIN           12 // MISO
    # define SPICLOCK         13 // sck
    # define VSYNC             2 // INT0
    # define MAX7456RESET      9 // RESET
    # define MAX7456SELECT    10 // CHIP SELECT 
    # define SETHARDWAREPORTS pinMode(MAX7456RESET,OUTPUT);pinMode(MAX7456SELECT,OUTPUT);pinMode(DATAOUT, OUTPUT);pinMode(DATAIN, INPUT);pinMode(SPICLOCK,OUTPUT);pinMode(VSYNC, INPUT);
    # define MAX7456HWRESET   digitalWrite(MAX7456RESET,LOW);delay(100);digitalWrite(MAX7456RESET,HIGH);
    # define MAX7456ENABLE    digitalWrite(MAX7456SELECT,LOW); 
    # define MAX7456DISABLE   digitalWrite(MAX7456SELECT,HIGH); 
#else                                  
    # define MAX7456ENABLE    PORTD&=B10111111; 
    # define MAX7456DISABLE   PORTD|=B01000000; 
    # define SETHARDWAREPORTS DDRB|=B00101100;DDRB&=B11101111;DDRD|=B01000000;DDRD&=B11111101;
    # define MAX7456HWRESET   PORTB&=B11111011;delay(100);PORTB|=B00000100;
#endif
```

dustin, 2016-07-08:
digitalWrite is insanely ugly and has an incredible amount of overhead.

jflyper, 2016-07-09:
Yes. It's a trade off between readability, portability, flexibility and extensibility v.s. efficiency.

That's the reason I asked if there's a substantial savings that justifies stashing all the goodness of digitalWrite() and pinMode().

tracernz, 2016-07-09:
The bitmasks could be built up from some nice defines, and the compiler will optimise it away anyway (although who knows with Arduino).

jflyper, 2016-07-10:
True. Something like digitalHigh() and digitalLow() would help, too. All we have to come up with is the digitalPinToPort() and digitalPinToBitMask() macros ... could be done with #if-#elif-#elif-#elif-...-#endif.

Does it worth the hassle?

jflyper, 2016-07-11:
Came up with something like this:
[wireMacro.c.txt](https://github.com/ShikOfTheRa/scarab-osd/files/356600/wireMacro.c.txt)

```
#define _BV(n) (1 << (n))

#define _dp2bm0  _BV(0) /* 0, port D */
#define _dp2bm1  _BV(1)
#define _dp2bm2  _BV(2)
#define _dp2bm3  _BV(3)
...
#define _dp2bm9  _BV(1)
...
#define _dp2bm16 _BV(2)
...
#define _digital_pin_to_bit_mask(n) _dp2bm##n

#define _dp2p0   D /* 0 */
#define _dp2p1   D
#define _dp2p2   D
#define _dp2p3   D
...
#define _dp2p9   B
...
#define _dp2p16  C
...

#define _dp2p(n) _dp2p ## n

#define _xxxconcat(a,b) a ## b
#define _digital_pin_to_xxx(reg, group) _xxxconcat(reg, group)
#define _digital_pin_to_ddr(n) _digital_pin_to_xxx(DDR, _dp2p(n))
#define _digital_pin_to_port(n) _digital_pin_to_xxx(PORT, _dp2p(n))

#define _pinModeOut(pin) _digital_pin_to_ddr(pin) |= _digital_pin_to_bit_mask(pin)
#define _pinModeIn(pin) _digital_pin_to_ddr(pin) &= ~_digital_pin_to_bit_mask(pin)
#define _digitalHi(pin) _digital_pin_to_port(pin) |= _digital_pin_to_bit_mask(pin)
#define _digitalLo(pin) _digital_pin_to_port(pin) & ~_digital_pin_to_bit_mask(pin)

doit()
{
   _pinModeIn(1);
   _pinModeOut(2);
   _digitalHi(3);
   _digitalLo(9);
   _digitalHi(16);
}
```
Ran it through gcc preprocessor:
```
$ gcc -E wireMacro.c
# 1 "wireMacro.c"
# 1 "<built-in>" 1
# 1 "<built-in>" 3
# 170 "<built-in>" 3
# 1 "<command line>" 1
# 1 "<built-in>" 2
# 1 "wireMacro.c" 2
# 59 "wireMacro.c"
doit()
{
   DDRD &= ~(1 << (1));
   DDRD |= (1 << (2));
   PORTD |= (1 << (3));
   PORTB & ~(1 << (1));
   PORTC |= (1 << (2));
}
$ 
```
Further work is required for simultaneous multi-port operation.

jflyper, 2016-07-11:
In reality, I/O port RMW should be done atomically, so
```
#define _digitalHi(pin) { \
    uint8_t _sreg = SREG; \
    cli(); \
    _digital_pin_to_port(pin) |= _digital_pin_to_bit_mask(pin); \
    SREG = _sreg;}
```
or similar is better.

tracernz, 2016-07-11:
Quite true, one of the reasons AVR is icky. I wonder if it could be nice to still have some way of doing a naked digitalHi/Lo so you can arrange them into blocks and avoid a little bit of overhead from multiple read/write/write SREG cycles. It's probably not that bad though I guess.

dustin, 2016-07-11:
That macro kind of reinvents ATOMIC_BLOCK.  I'd recommend just using the
built-in:
http://www.nongnu.org/avr-libc/user-manual/group__util__atomic.html

jflyper, 2016-07-12:
@dustin Thanks! I didn't know that!

jflyper, 2016-07-12:
Handling multiple pins all at the same time.
By encoding port and bit for a pin into a single macro variable, we can come up with something like this:
```
#define _dp2xbm0  (_BV(0) << 16) /* 0, port D */
#define _dp2xbm1  (_BV(1) << 16)
#define _dp2xbm2  (_BV(2) << 16)
#define _dp2xbm3  (_BV(3) << 16)
...
#define _dp2xbm8  _BV(0) /* 8, port B */
#define _dp2xbm9  _BV(1)
#define _dp2xbm10 _BV(2)
#define _dp2xbm11 _BV(3)
...
#define _dp2xbm14 (_BV(0) << 8) /* 14, port C */
#define _dp2xbm15 (_BV(1) << 8)
#define _dp2xbm16 (_BV(2) << 8)
#define _dp2xbm17 (_BV(3) << 8)
...
#define _digital_pin_to_extended_bit_mask(n) _dp2xbm ## n
#define _xm_(n) _digital_pin_to_extended_bit_mask(n) // short hand
```
Then we can define
```
// Set OUT direction for multiple pins
#define _pinModeOutGroup(bm) ATOMIC_BLOCK(ATOMIC_RESTORE_STATE){ \
  if ((bm) & 0xff) { DDRB |= ((bm) & 0xff); } \
  if (((bm) >> 8) & 0xff) { DDRC |= (((bm) >> 8) & 0xff); } \
  if (((bm) >> 16) & 0xff) { DDRD |= (((bm) >> 16) & 0xff); } }

// Set IN direction for multiple pins
#define _pinModeInGroup(bm) ATOMIC_BLOCK(ATOMIC_RESTORE_STATE){ \
  if ((bm) & 0xff) { DDRB &= ~((bm) & 0xff); } \
  if (((bm) >> 8) & 0xff) { DDRC &= ~(((bm) >> 8) & 0xff); } \
  if (((bm) >> 16) & 0xff) { DDRD &= ~(((bm) >> 16) & 0xff); } }

// Set IN and OUT pins simultaneously
#define _pinModeIO(ibm, obm) ATOMIC_BLOCK(ATOMIC_RESTORE_STATE){ \
  DDRB = (DDRB & ~((ibm) & 0xff)) | ((obm) & 0xff);\
  DDRC = (DDRC & ~(((ibm) >> 8) & 0xff)) | (((obm) >> 8) & 0xff);\
  DDRD = (DDRD & ~(((ibm) >> 16) & 0xff)) | (((obm) >> 16) & 0xff); }
```

Test calls to these macros,
```
_pinModeOutGroup(_xm_(2)|_xm_(3)|_xm_(11)|_xm_(14)|_xm_(15));
_pinModeIO(_xm_(2)|_xm_(11)|_xm_(14), _xm_(3)|_xm_(15));
```
when compiled with Xcode native gcc, it looks like the gcc can delete the if-body for false cases.

Gonna test with Arduino.

jflyper, 2016-07-14:
Hey, folks, I was comparing the actual output from the hard coded v.s. macro based code, and found out that
```
 # define SETHARDWAREPORTS DDRB|=B00101100;DDRB&=B11101111;DDRD|=B01000000;DDRD&=B11111101;
```
is actually incorrect; the last bit clear operation against DDRD is for VSYNC (pin 2) input, and the mask should be B11111011, not B11111101.

Final results do not differ, but since the mask B11111101 is for pin 1 (RXI), @ShikOfTheRa was just lucky.

ShikOfTheRa, 2016-07-21:
Yep - lucky :) 
PR's applied.

closed 2016-07-26 by ShikOfTheRa