ShikOfTheRa/scarab-osd #284: OSD working, no video feed (gray background)

eluminer, 2016-07-16:
Hello guys!

**Problem:**
I have working OSD (voltage, RSSI, current from FC) but **no video.**

**Components:**
Micro MinimOSD
X-Racer F303 V2.1
TS58500 VTx
RunCam Swift (PAL)

**Behaviour:**
On my monitor I can see the OSD, enter the PID menu, change stuff, see correct voltage and everything. However, I do not see the video from the camera.

When I connect the camera directly to the VTx then I have the video (no OSD, of course).

**What I have tried so far:**
All connections have common ground (checked 10 times)
I've changed the camera and the VTx to PZ0402 and TS5828
I've change PAL/NTSC setting in GUI and used AUTOCAM feature in config.h
Tried to play with VSYNC, FASTPIXEL, WHITE and BLACK BRIGHTNESS
Changed the OSD
Changed settings on the camera from CRT to LCD
Tried versions 1.5 and 1.6.1

I am hopeless. I just can't get the video throught the OSD.

Please help!

Dean

jflyper, 2016-07-16:
Looks to me that you have a MAX chip with bad video input side.

Why don't you try the development version (goto the code tab, select master branch and download), and tell us what the last line of the opening screen say.

eluminer, 2016-07-16:
SI: Not detected - (NTSC or PAL, whatever I select in GUI)

eluminer, 2016-07-16:
Does "bad video input side" means hardware issue or? Because I have tried 2 brand new Micro MinimOSDs and the same thing happens. Also, 2 different cameras where also tried.

jflyper, 2016-07-17:
Oops, I noticed that the master has some problem.

Can you try downloading from this version and try again? Sorry for your inconvenience.
[https://github.com/jflyper/scarab-osd/tree/jflyper-ahead](https://github.com/jflyper/scarab-osd/tree/jflyper-ahead)

eluminer, 2016-07-17:
I tried this one. Same thing. Only thing I change is define micro minimosd. The rest I leave default.

I still says SI: not detected.

:(

eluminer, 2016-07-17:
I've tried to disable AUTOCAM (selecting both NTSC and PAL), disabling VSYNC. Still no video! :(

jflyper, 2016-07-17:
Hmmm... There's not much we can do from the software side at this point.

Check CAM-OSD wiring one last time, may be noting VIN-GND-GND-VOUT order, then...
Get another micro or two from a different source, may be?

eluminer, 2016-07-17:
I've ordered 2 more OSDs from different source. I will rewire everything from the beginning today and try again. 

Will let you know if I make a progress.

Thank you very much for quick support.

eluminer, 2016-07-17:
OMG!!!!!!!!!

I'm kinda embarrassed but I have to share the solution. So, nothing software related.

The cheap wire that I was using to connect camera to OSD was broken inside itself in the middle. There was no externally visible damage to the insulation, but when I (today) checked the wire for continuity I couldn't believe that multimeter doesn't beep!!!

When I was changing the OSDs I used the same wire in both cases, that is why changing the OSD didn't solve the problem. And when I was connecting the camera to the VTx directly I used different wire. :D

I'm happy and mad at the same time! :/

Thank you very much for the help jflyper!

Best regards!

closed 2016-07-17 by eluminer

jflyper, 2016-07-17:
SH*T happens!!! 😉