ShikOfTheRa/scarab-osd #286: AB7456, twice the character memory?

jflyper, 2016-07-30:
There has been a rumor sometime that AB7456 has what so called twice the character memory, but airbot officially(?) confirmed it with the release of the MicroOSD v2.3. [http://shop.myairbot.com/index.php/microosdv23.html](http://shop.myairbot.com/index.php/microosdv23.html)

So, what is the character memory anyways? Font glyph memory? Video memory? How do we access them? What are the great things that can be done with them?

nathantsoi, 2016-07-30:
i think it's the nvram. i asked him how we can tell if the device has 2x
the nvram, but it doesnt sound like this is possible in software. we might
try just accessing the next block and seeing if it fails?

could be cool for logos and stuff :)

jflyper, 2016-07-30:
@nathantsoi Yeah, the current MAX design already has entire 8-bit addressable character memory, so there must be some control bit to enable access to the next block. So, we need to see some reference manual.

jflyper, 2016-07-30:
There's more to find out about the AB7456. In the myairbot page, AB7456 is said to be operate at 3.3V.

ShikOfTheRa, 2016-08-04:
Have asked Li at airbot... if he has the info

jflyper, 2016-08-27:
myairbot's Micro OSD v2.3.
I had it for a couple of weeks, but haven't get a chance to give it a try.
This morning, I gave it a try. Just hooked it up to FTDI and GUI, no VID connection. Also did some visual inspection.
Here are the current findings.

1. Runs _**much^2**_ cooler than MAX-based implementation.
Thanks to AB7456 running at 3V3?
2. Sometimes (after flashing, to be precise) fail to reset AB7456 properly, breathing SCLK.
Different reset characteristics/timings?
3. HSYNC wired instead of VSYNC? (Need verification)
Somebody in OMNIBUS F3 thread in RCG told me that AB7456 can handle screen memory writes in arbitral timing without flickering; suggests the AB7456 has a dual port screen memory or an ability to buffer the writes until next HVSYNC (which I doubt it has either).
If HSYNC is actually wired instead of VSYNC, we may have to review the burst write timings.

We really need a descent data sheet.

ShikOfTheRa, 2016-08-28:
I fed back same to airbot. They have requested data sheet from factory.

Rodrigoah, 2016-08-31:
I have mine working, I am willing to do some research. 

I had to set the board type to minimosd instead of Micro MinimOSD, took me a while to figure out...

ShikOfTheRa, 2016-09-14:
IS it worth keeping this open? No coding to do...

jflyper, 2016-09-14:
Let's close it.

closed 2016-09-14 by ShikOfTheRa