ShikOfTheRa/scarab-osd #290: MWOSD GUI - RC switch channel limited to 8CH (0-7)

csurf, 2016-08-11:
The GUI needs to be updated to allow for picking higher channels for the RC switch channels above the 0-7 range. It should at least go up to 16CH (values 0-15).

BTW, I tried to compile the GUI myself, with processing 2.2.1 and controlP5 2.1.5. It errored out during the compile. Is the GUI code currently broken?

jflyper, 2016-08-15:
> Side-note, it would be nice to change the channel indexing to start at 1 instead of being zero-based indexing; seems to make more sense this way.

+1

ShikOfTheRa, 2016-08-18:
Added to the roadmap.

ShikOfTheRa, 2016-09-06:
Amended for ch1-8

To implement this its unfortunate that MSP_RC doesn't say how many channels - so either have to work it out by counting bytes (meh...) or have a fixed 8/16 option that can select in config.h. :) Does that work?

ShikOfTheRa, 2016-09-06:
Well added something for testing. I don't have means to test...

My tx is 22 years old. lol

closed 2016-09-14 by ShikOfTheRa

jflyper, 2016-09-24:
I saw the mod. Doesn't the index shifting confuse other part of the code?

ShikOfTheRa, 2016-09-24:
I think I just changed the displayed value. That was the intent. Otherwise it hats a bit complicated across all the different options

jflyper, 2016-09-26:
IIRC, I've seen MwRc{something} read into positions 1,2,3,4,... instead of 0,1,2,3... Let me check.

ShikOfTheRa, 2016-09-26:
I think on gui just make it 1-8.
On osd I extended mwrcdata array by one and changed all the channel numbers to match. In serial reads into mwrcdata starting at one. Am travelling and on mobile so can't check. Not back for another week. I have used it since so it should be ok..