ShikOfTheRa/scarab-osd #291: AH behavior on extreme pitch

jflyper, 2016-08-11:
AH tilts to left (indicates right roll) on extreme up or down pitch.

jflyper, 2016-08-11:
Okay, FC is sending these angles. The phenomenon begins at around 45~50 degree pitch.
I don't know if they are legitimate values, but look bad when displayed as AH.
Interestingly, CFC's acc scope shows no such values.

jflyper, 2016-08-16:
So the roll doesn't mean much if the pitch is extreme.

I wonder how digital HUD/AHI in a real aircraft handle the situation.

tracernz, 2016-08-16:
Google "Euler angles singularity". The solution is quaternions.

jflyper, 2016-08-16:
Yeah, I was aware of the singularity problem, but didn't know the quaternions. Gees, it's math!

I'm noob with attitude control code; do uav flight control use quaternions?

tracernz, 2016-08-16:
Multiwii and derivatives (AFAIK) don't. OpenPilot and derivatives do. I imagine PX4 does too.

DzikuVx, 2016-08-16:
There is an idea to rewrite INAV to quaternions but it's only an idea so far

ShikOfTheRa, 2016-09-14:
Not sure much we can do her ewith MWOSD code - OK to close?

as fyi, the following constraints are available  in config.h not that  its a lot of use!

#define AHIPITCHMAX 200             // Specify maximum AHI pitch value displayed. Default 200 = 20.0 degrees
#define AHIROLLMAX  400             // Specify maximum AHI roll value displayed. Default 400 = 40.0 degrees

jflyper, 2016-09-14:
Yeah, I will keep it as yet another thing to care (once in a while) list.

closed 2016-09-14 by ShikOfTheRa