ShikOfTheRa/scarab-osd #292: Support for APM/MAVLink wont work

mheid, 2016-08-13:
Hi,
i tested the APM/MAVLink support and it currently wont work.

jflyper, 2016-08-13:
I believe it's still in development (at far as a comment in Config.h tells me).

I confirmed the basic operation (it can receive MAVLINK telemetry messages), but noticed that attitude (AHI) is too sensitive (incorrectly scaled).

What is your "won't work" status? Please elaborate.

jflyper, 2016-08-13:
AHI scaling fix PR has been submitted (#295)

mheid, 2016-08-13:
The overlay shows "no data" but the RX of the OSD is receiving and telemetry output still works.
Other/older forks of the minimosd firmware working fine, but i really like to use this one with the mac compatible GUI.
Which config parameters did you use for getting the basic operations up and running?
Maybe i did mistakes by selecting the right parameters in config.h

jflyper, 2016-08-13:
I tested with CleanFlight 1.13 (current master), MWOSD current master; just turned on PROTOCOL_MAVLINK in Config.h.
(With AHI scaling fix, you should consult [https://github.com/jflyper/scarab-osd/tree/fix-MAVLINK-attitude-calculation](https://github.com/jflyper/scarab-osd/tree/fix-MAVLINK-attitude-calculation)).

NO DATA means the MWOSD is not receiving the valid data over the comm line.

How about reviewing the followings?

0. RX/TX cross over. Since it's a telemetry, only RX is important.
1. FC is not set to emit MAVLINK telemetry messages.
2. Serial speeds do not agree. If you are using CF and left telemetry speed to AUTO, it will default to 57600 bps and you should modify BAUDRATE define in Config.h also to match the speed.

mheid, 2016-08-13:
Here we have the difference, i tested with APM as the title mentioned.
The code mentioned "Requires testing" regarding the APM compatibility but i think right now its not given / broken.
The same wiring of all the components works fine with other/older versions of the minimosd firmware.
I just roughly tried different config settings for the apm but i did not deep dive into the code.

jflyper, 2016-08-13:
I suspect the serial speed then. Did you check?

The MAVLINK support in CF is around for a while and (I suppose) it has been used by people with APM/MAVLINK OSD, so it's hard to imagine the CF support is badly broken.

mheid, 2016-08-13:
Yep, i tried different baud-rates just to be sure, i also tried different config settings.
You mentioned cleanflight again.. just to be sure: I only tested with APM 2.6 & 2.8 Hardware with arduplane installed on it.

jflyper, 2016-08-13:
Which firmware did you try? The current master? Or R1.6.1.0?

mheid, 2016-08-13:
To be honest, i already deleted the folder with all the versions i tested.
I will test master and R1.6.1.0 tomorrow again and will let you know status for both versions.
Are they any other informations i can collect while the test for debugging which will help?

jflyper, 2016-08-14:
If you have a spare time, can you try the version that displays debug information about serial message parsing?
[https://github.com/jflyper/scarab-osd/tree/mavlink-debug1](https://github.com/jflyper/scarab-osd/tree/mavlink-debug1)

It already has TELEMETRY_MAVLINK and BAUDRATE set to 57600.

mheid, 2016-08-14:
Sure, i will let you know as soon i tested.

coolest42u, 2016-08-31:
hi,
i am using micro minimOSD and miniAPM. When i connect  apm to mission planner by data link and forward the mavlink data from mission planner  to OSD, everything works. But When i  directly connect osd to apm telemtry port it doesnt work. Only shows some symbols on the screen. I try to change telemetry delay to 5 sec, nothing change. May be Mavlink protocol didn't match. (0.9 or 1.0) I think APM use Mavlink ver. 0.9 Pixhavk use Mavlink ver. 1.0

jflyper, 2016-08-31:
Which OSD software are you using?

coolest42u, 2016-08-31:
scarab-osd-1.6.1.0

coolest42u, 2016-08-31:
i attached my config.h as txt

[Config.txt](https://github.com/ShikOfTheRa/scarab-osd/files/447305/Config.txt)

jflyper, 2016-08-31:
Okay... I believe MAVLINK is still under development and/or haven't been tested throughly.
Let's wait @ShikOfTheRa to chime in.

coolest42u, 2016-08-31:
thnx

coolest42u, 2016-09-13:
Hi, 
The new beta still don't work. Other firmwares (minimOSD extra and MinimOSD night Ghost) are works with the same hardware setup. I don't understand why it works port forwarding from Mission Planner to minimOSD  (APM 2.5-->Datalink-->Mission Planner -->Forward Serial Port -->USB to TTL-->minimOSD) , but not work stand alone APM 2.5 telemetry port?

ShikOfTheRa, 2016-09-13:
Hi - I'm not too familiar with the different APM / pixhawk flavours etc. so have done as best that I can with little knowledge.
I suspect MWOSD is not requesting the correct info to be sent to OSD - are you able to set it up so it sends automatically? I started this before ghost came on line with his changes for the APM OSD and maybe that is the better long term option. I would like to conclude this, but think its a few weeks before I can look at this. I have an APM board - just need a copter to fit it into!

coolest42u, 2016-09-13:
hi thank you for your reply, "are you able to set it up so it sends automatically?" do you mean "#define MAVLINKREQ" if so, yes I enabled it

ShikOfTheRa, 2016-09-13:
Sorry - I mean in APM somewhere - so it always send. Like telemetry. Also it is version 1.0 - that may make a difference in some cases

coolest42u, 2016-09-13:
ok I will try to find this parameter in APM configuration file

coolest42u, 2016-09-13:
No luck, I think it's not APM dependent because other firmwares works with the same APM config and hardware. ı change the firmware everything works but not MW firmware.
Anyway thank you for your interest

coolest42u, 2016-09-13:
On pixhawk everything works :) I think mavlink version on APM wrong....

ShikOfTheRa, 2016-09-13:
Thats brill. Thanks for feedback. I dont use apm / pixhawk so this is a bit of an unknown for me. If find something not working please post and I'll see if can resolve

coolest42u, 2016-09-13:
I think I am very close to solution on APM. Now I can compare the configuration of them. I will write here the solve :)

ShikOfTheRa, 2016-10-06:
@coolest42u how did your testing go:

closed 2016-10-20 by ShikOfTheRa

ShikOfTheRa, 2016-10-31:
@coolest42u did you have any luck with mwosd on mavlink 0.9 ?