ShikOfTheRa/scarab-osd #293: MSP amperage is signed 16-bit

jflyper, 2016-08-13:
MSP amperage is signed 16-bit integer (unless MultiWii mode is selected).

[https://github.com/cleanflight/cleanflight/blob/v1.13.0/src/main/io/msp.c#L677-L681](https://github.com/cleanflight/cleanflight/blob/v1.13.0/src/main/io/msp.c#L677-L681)
```
            if(batteryConfig()->multiwiiCurrentMeterOutput) {
                sbufWriteU16(dst, (uint16_t)constrain(amperage * 10, 0, 0xFFFF)); // send amperage in 0.001 A steps. Negative range is truncated to zero
            } else
                sbufWriteU16(dst, (int16_t)constrain(amperage, -0x8000, 0x7FFF)); // send amperage in 0.01 A steps, range is -320A to 320A
            break;
```

Current MWOSD code interprets the number as unsigned and results in meaningless display if negative amperage is given.

ShikOfTheRa, 2016-08-18:
Do we get negative amperage ever? unless bad config set. IS there some new regenerative ESC or something.
Or maybe current sensor with negative offerset?

jflyper, 2016-08-18:
The PR was motivated by a user complaining the negative amperage generated by BF (probably too much offset) being displayed as lightning level figure.

As the feature is "Use FC Amperage", I thought it would be more legit to display what FC tells OSD to display.

In practice, I believe vast majority of current sensors are uni-directional (those based on INA-169), and should never give negative values.

closed 2016-08-18 by jflyper

ShikOfTheRa, 2016-08-19:
OK. we should cater for less than perfect scenarios