ShikOfTheRa/scarab-osd #296: Popular boards hex files on release/pre-release

slacker87, 2016-08-14:
Any way we can get precompiled hex files of the releases and pre-releases for the main boards supported? (minimosd, microminimosd, etc)

I'm beginning work on a configurator for MWOSD and having everything in one place would be appreciated.

https://github.com/slacker87/MWOSD-Configuration-Tool

dustin, 2016-08-14:
Much of the configuration is done as part of compilation.  A better
strategy would be to include that part of configuration in a config tool.
It's a little outside of the scope of a typical chrome app that only talks
over serial, but it would be very, very nice to have a compile server
sitting somewhere doing that work.

This is a bit harder, but not impossible with the Arduino base.

slacker87, 2016-08-14:
The arduino IDE supports saving sketches as a hex file. I was simply seeing if it was easy enough to export and upload that as part of the main releases, if not then yes either a manual or automated compile can be done.

dustin, 2016-08-14:
Consider this:
https://github.com/ShikOfTheRa/scarab-osd/blob/master/MW_OSD/Config.h

slacker87, 2016-08-14:
Correct, your recommending compilation on the fly from chrome?

dustin, 2016-08-14:
That's the only feasible way to do it.  Either in javascript, or via a
cloud service.

ShikOfTheRa, 2016-08-18:
Actually, there is a gui configurator due for release shortly.

Can program arduino direct from chrome. I think it's not difficult. I went so far but remember struggling with knowing memory location for boot loader vs code etc. I read arduino image out into a binary file and verified but never bothered going past that. Am sure someone with some time could easily complete that.

slacker87, 2016-08-18:
link to source of gui configurator?

ShikOfTheRa, 2016-08-19:
Unfortunately its not going to be open as its written from ground up and not by me.
I probably still have the arduino programming stuff somewhere if interested.

slacker87, 2016-08-19:
Absolutely!

Also closed source, booo....

ShikOfTheRa, 2016-08-22:
OK - will try to find it today. 
closed- yeah I know, but was part of the deal I committed too.

With google shutting own chrome support it EOL already!! Damn I hate google - that the third time in 18 months they have pulled the plug on something used a lot.

tracernz, 2016-08-22:
The writing was kind of on the wall. ChromeOS has enjoyed far better app API support for a long time.

ShikOfTheRa, 2016-08-22:
Yes - Google are doing a lot of cutting lately. 
Really dissapointed - moved 4500 users to office365 because of their recent attitude....

ShikOfTheRa, 2016-08-24:
Any thoughts on where people will go now for a cross platform tool to replace current for CF/BF configurators etc.

slacker87, 2016-08-24:
I've been looking to electron, it works really well in my testing.

ShikOfTheRa, 2016-08-24:
How easy do you think it will be for a novice to install and run an app. Some of the installs look a little ugly to me - having to run cli commands to get things running. It kinda put me off. Developer side OK...

slacker87, 2016-08-24:
Dev side when you release a new version, you use electron-packager and it gets converted into a .exe/.app executable file. So the end user only needs to run said exe or app. 

https://www.npmjs.com/package/electron-packager

ShikOfTheRa, 2016-08-24:
That's how I'd like to think it would work. Have to try it out... Its definitely a contender. How about QT?

Tomcatt, 2016-08-24:
@ShikOfTheRa Are you looking to get updated links for the Configurators?  I found some github links that I download and point chrome to that directory.

tracernz, 2016-08-24:
You could do something pretty good with Qt's QML and Javascript if you want to stick with JS. You also have the power of writing some portions in C++ if needed.

closed 2016-09-14 by ShikOfTheRa